steveicarus/iverilog #166: Does icarus support SV virtual interface?

svenka3, 2017-10-04:
Am trying to port some small SystemVerilog training lab examples to run on Icarus. So far it has been good experience. Am able to run small SV-Design constructs with small tweaks. So kudos to the -g2012 team!

Am getting syntax error for "virtual interface inside a class". Is this known issue? if yes, is there an alternate way you recommend to connect SV TB to RTL?

Thanks
Srini

martinwhitaker, 2017-10-07:
I don't know what you mean by "virtual interface inside a class". Could you provide a simple example.

svenka3, 2017-10-07:
Consider an example 

interface vlb_if;
  logic vlb_wr;
endinterface: vlb_if

class vlb_drvr;
  virtual vlb_if vif;
endclass 

Thanks 
Srini

martinwhitaker, 2017-10-07:
Sorry, no, Icarus does not currently support interfaces in any useful way. It lets you define them, but that's all. Adding support for this would be a major piece of work.

fl4shk, 2018-06-11:
I'm kind of interested in contributing to Icarus Verilog specifically on this issue.  How hard would it be to add in support for an interface that simply consists of inputs, outputs, and parameters?  Support for interfaces with only as much functionality as this would be really nice:

```
interface Example #(parameter N = 8)
  (input logic clk, rst_n, input logic [N_MINUS_1 : 0] in_data, 
  output logic [N_MINUS_1 : 0] out_data);
localparam N_MINUS_1 = N - 1;
endinterface
```

fl4shk, 2018-06-12:
I'm going to go ahead and open a new issue for this.

jotego, 2020-11-29:
I would use interfaces on module ports if available. 
For a module use only case, implementation cannot be that hard. It just seems like joining strings when forming signal names internally. 

Thank you

holodnak, 2022-08-19:
Any progress on this?

caryr, 2024-01-23:
Here is another example of code already in the tests: ivtest/ivltests/sv_interface.v