steveicarus/iverilog #167: SV class inheritance - not working?

svenka3, 2017-10-04:
Hello,
   I am trying a simple SV class code with IVL. Simple class works. Inheritance doesn't compile. I get an error as __../ref/inherit_base.sv:51: assert: elab_scope.cc:503: failed assertion scope__

I am using __-g2012__ already. Am I missing a flag or something please?

Thanks
Srini 

Consider the following code:

```javascript
class s2p_pkt;
 bit [15:0] hdr= 16'd10;
 bit [15:0] pkt_pld=16'd30;

  virtual function bit [31:0] bit_pack();
      bit_pack = {hdr,pkt_pld};
  endfunction : bit_pack

  function void display ();
    $display ("CVC:VSV: Lab3: Learning Inheritence");
    $display ("CVC:VSV: Lab3: Extended class %m has base properties hdr %0b pkt_pld %0b AND Extra err_pkt ", hdr, pkt_pld );
  endfunction : display

endclass : s2p_pkt

class s2p_err_pkt extends s2p_pkt;
endless

```

martinwhitaker, 2017-10-07:
Your example is incomplete, and ends with an invalid keyword.

svenka3, 2017-10-07:
I believe I had a typo in last line 

Just replace it:

endclass 

That should be complete 

Thanks 
Srini

martinwhitaker, 2017-10-07:
I tried that, and it compiled without error.

There are only 17 lines, but the error message you reported is for line 51...

svenka3, 2017-10-07:
Interesting. Let me post the original code tomorrow 

Thanks for prompt reply

matthamptonasic, 2017-10-07:
I have an open issue here where iVerilog core dumps if all classes are in the root scope.  Maybe that could be the difference if Martin is compiling within a program block and Srini is not?
[https://github.com/steveicarus/iverilog/issues/98](url)

svenka3, 2017-10-07:
That is certainly my use case. Thanks Matt. Will try with a program block soon

Regards 
Srini

martinwhitaker, 2017-10-07:
I can reproduce the fault by adding code to instantiate an object of the derived class. It looks like Matt is correct and this is a duplicate of issue 98.

martinwhitaker, 2017-10-08:
This should now be fixed in both the master and v10 branches.

closed 2017-10-08 by martinwhitaker