steveicarus/iverilog #169: $dumpvars of a signal that goes through the resolve list ends up with wrong scope

abottelli, 2017-11-04:
Everyone,
I've been working on a fairly big design and I've been using iverilog and vvp to simulate it.
I've recently switched to version 10 to make use of the support for multi-dimensional arrays, but I noticed an issue on the scope properties of signals dumped by $dumpvars directives. In practice, some signals (not all), ended up with a totally wrong scope in the header of the generated VCD file.
So, I decided to spend some time debugging the problem and I came up with a possible cause and solution for it. Here's a description, sorry if some of the details are not exactly right, but I looked at the code for the first time only a few days ago.

The vvp code maintains a data structure (special vpiHandle called __vpiScope) that has information about each scope. As part of that class, there is a vector (intern) that is supposed to
maintain a list of all the signals that are part of that scope (through special vpiHandle called __vpiSignal). Each signal's vpiHandle though has a "within" pointer that is supposed to point to the scope handle. This way you can get to the signal scope using the vpiHandle
of the signal, which is what the $dumpvars directive does. 
When a net is compiled, the vpiHandle of the signal is created and added to the vector (intern) of the scope, but the pointer for each vpiHandle to its scope is not updated. It happens that the vpiHandle points to the current scope instead.
When the design is compiled, during the first pass, and a net is compiled, it just happens that the current scope is also the correct scope for the signal, so the problem doesn't show.
When instead a net goes through the "resolve list" of issues to resolve, the current scope now does not match and the issue above appears.
The original scope of the signal is available to the do_compile_net function, so to solve the issue I just set the signal vpiHandle pointer to the correct scope. What makes it slightly more complicated is that only the vpiHandle for signals (__vpiSignal) has the scope pointer, so I had to do a dynamic cast, which I have seen doing in other parts of the code.

I don't know if the scope of each signal vpiHandle happens to be used only for $dumpvars directives and nothing more. Moreover, the issue appears only on nets that go through the resolve_list. This is maybe why nobody has seen the issue so far, but in our design this happens for thousands of signals, so maybe I'm not seeing the complete picture here.

I have a commit that solves the issue for you to review, but the problem might involve more cases (for example, the code for real variables seems pretty similar).

Thank you,
Aldo Bottelli

martinwhitaker, 2017-11-07:
Thanks for the report and the detailed analysis! You are right that the same problem affects real nets (variables aren't affected because they can always be resolved immediately).

I've chosen to fix this a little differently, by passing down the correct scope to the functions that fill in the __vpiSignal objects. Fix applied to both the master and v10 branches. I've added a pair of regressions tests to the test suite which should demonstrate the problem is fixed, but please confirm this on your larger design.

abottelli, 2017-11-08:
I confirm that the problem is resolved. Thank you.

closed 2017-11-08 by abottelli