steveicarus/iverilog #171: output is not realtime when using pipe or redirection

sgherbst, 2017-11-06:
Hi,

It seems that **$display** messages don't appear until the end of simulation when using a pipe or redirection on the **vvp** output.  For me this is mainly an issue when monitoring the **vvp** output via a Python script, but I can also reproduce the issue using **tee** or output redirection.

In each case, realtime behavior is restored by calling **vvp** through **unbuffer**.  For reference, here is [a.out](https://gist.github.com/sgherbst/6a4bb1b9bdaf8da327a5cacc1fcaa704#file-a-out) from my machine.

setup for tests ([test.v](https://gist.github.com/sgherbst/d755d257f20fb1df2382a968c4637c29#file-test-v))
`iverilog -o a.out test.v`

example 1: nothing printed to test.txt until the end of the simulation
`./a.out > test.txt`
fixed by using unbuffer
`unbuffer ./a.out > test.txt`

example 2: nothing printed to stdout or test.txt until the end of the simulation
`./a.out | tee test.txt`
fixed by using unbuffer
`unbuffer ./a.out | tee test.txt`

example 3: monitoring via python ([test.py](https://gist.github.com/sgherbst/6488c780256fca3b31e8df7d552f2841#file-test-py))
`python3 test.py`
fixed by using unbuffer ([test2.py](https://gist.github.com/sgherbst/7c2d75ad02d57f7f32ecfd2886acb5bb#file-test2-py))
`python3 test2.py`

Setup:
macOS Sierra 10.12.5
vvp 11.0
iverilog 11.0

Thanks,
Steven

martinwhitaker, 2017-11-06:
This is normal behaviour for standard output on a Unix-like system. printf in a 'C' program gives the same behaviour. Why would you expect anything different?

sgherbst, 2017-11-06:
Thanks for the fast reply!  If this is the desired behavior then I'm fine to close it.

closed 2017-11-06 by sgherbst