steveicarus/iverilog #172: Error during compile

red0bear, 2017-11-08:
This error appear when i try compile with last updates.

flex -s -t lexor.lex > lexor.cc
bison --verbose -t -p VL -d -o parse.cc parse.y
parse.y:699.1-11: error: diretiva inválida: `%precedence'
parse.y:700.1-11: error: diretiva inválida: `%precedence'
parse.y:701.1-11: error: diretiva inválida: `%precedence'
Makefile:257: recipe for target 'parse.cc' failed

martinwhitaker, 2017-11-08:
On 08/11/17 09:53, Felipe Fernandes da Costa wrote:
> This error appear when i try compile with last updates.
>
> flex -s -t lexor.lex > lexor.cc
> bison --verbose -t -p VL -d -o parse.cc parse.y
> parse.y:699.1-11: error: diretiva inválida: `%precedence' [...]

What version of bison are you using?

red0bear, 2017-11-08:
bison 2.7-11.15

martinwhitaker, 2017-11-08:
On 08/11/17 12:51, Felipe Fernandes da Costa wrote:
> bison 2.7-11.15
> 
Strange, according to the change log, %precedence has been supported since bison 2.4.1.

Try changing the %precedence directives to %nonassoc.

red0bear, 2017-11-08:
Works, but here more info:

line 698 in parse.y

 /* to resolve timeunits declaration/redeclaration ambiguity */
//%precedence no_timeunits_declaration
//%precedence one_timeunits_declaration
//%precedence K_timeunit K_timeprecision
%nonassoc no_timeunits_declaration
%nonassoc one_timeunits_declaration
%nonassoc K_timeunit K_timeprecision

this is other output warnings during compile:

g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow   -g -O2 -MD -c lexor.cc -o lexor.o
<stdout>: In function ‘int yy_get_next_buffer()’:
<stdout>:3005:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
<stdout>:1327:20: note: in definition of macro ‘YY_INPUT’
<stdout>: In function ‘yy_buffer_state* VL_scan_bytes(const char*, yy_size_t)’:
<stdout>:3507:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

<stdout>: In function ‘yy_get_next_buffer’:
<stdout>:625:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
<stdout>:1083:3: note: in expansion of macro ‘YY_INPUT’
<stdout>: In function ‘readmem_scan_bytes’:
<stdout>:1588:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

martinwhitaker, 2017-11-08:
OK, I've made the change to %nonassoc.

For the other warnings you are seeing, they are coming from code generated by flex. I suspect that means you are using an old version of flex as well.

closed 2017-11-08 by martinwhitaker

red0bear, 2017-11-08:
i just installed the default from last opensuse ... i dont know how 2.7 could turn 2.4 .... anyway thanks