steveicarus/iverilog #173: Possibly incorrect "duplicate declaration for net or variable" message

carlodri, 2017-11-09:
I have a Verilog file where `iverilog` reports multiple "duplicate declaration for net or variable" errors. However, the design in Quartus compiles correctly. I'm almost sure that the syntax is valid Verilog, and I cannot reproduce the error in a MWE. Since I would not like to publish the entire file here publicly, can confidentially send the file to @steveicarus  in order to diagnose what's happening?

red0bear, 2017-11-09:
Do you know quartus or more new tools remove another duplicated net while compile. Icarus could do it too ?

carlodri, 2017-11-09:
Just to clarify, this is the _kind_ of duplication `iverilog` is complaining about:

```verilog
module something (input mywire);
endmodule // something

module something_2 (output reg mywire);
endmodule // something

module top_module ();
  wire mywire;
  something something_0(
    .mywire(mywire)
    );
  something_2 something_2_0(
    .mywire(mywire)
    );
endmodule // top_module
```

(he complains about `mywire`) but, as I said, in this small example he does _not_ complain.

martinwhitaker, 2017-11-09:
Check you haven't inadvertently used a variable before declaring it, e.g.

```verilog
module something (input mywire);
endmodule // something

module something_2 (output reg mywire);
endmodule // something

module top_module ();
    something something_0(
      .mywire(mywire)
      );

    wire mywire;

    something_2 something_2_0(
      .mywire(mywire)
      );
endmodule // top_module
```

In this example, the first use will result in an implicit wire declaration, so the following 
explicit declaration will be reported as a duplicate.

carlodri, 2017-11-13:
ok @martinwhitaker that is indeed the case, thank you!

closed 2017-11-13 by carlodri