steveicarus/iverilog #174: Value got in same edge are different

red0bear, 2017-11-13:
hi there, 

I'm facing a little problem here when try to get a value from a edge signal where the same register have different value when read in same edge. The most strange behavior i have another call for get data and it doesn't happen in control. I will try run in another simulation tool with some result. 

Ex not working  working cause bit_c_3;  is different when is got from control and and parity:

always@(posedge ready_control or negedge rx_resetn )
begin
	if(!rx_resetn)
	begin
		control_r	   	<= 4'd0;
		parity_rec_c 	  	<= 1'b0;
		parity_rec_c_gen 	<= 1'b0;
	end
	else
	begin

		control_r	  <= {bit_c_3,bit_c_2,bit_c_1,bit_c_0};
		parity_rec_c	  <= bit_c_3;

		if(last_is_control)
		begin
			parity_rec_c_gen <= !(bit_c_2^control[0]^control[1]);
		end
		else if(last_is_data)
		begin
			parity_rec_c_gen <= !(bit_c_2^data[7]^data[6]^data[5]^data[4]^data[3]^data[2]^data[1]^data[0]);
		end
	end
end 

Another on working well during simulation

always@(posedge ready_data or negedge rx_resetn )
begin
	if(!rx_resetn)
	begin
		dta_timec	   	<= 10'd0;
		parity_rec_d 	  	<= 1'b0;
		parity_rec_d_gen 	<= 1'b0;
	end
	else
	begin
		dta_timec	  <= {bit_d_9,bit_d_8,bit_d_0,bit_d_1,bit_d_2,bit_d_3,bit_d_4,bit_d_5,bit_d_6,bit_d_7};
		parity_rec_d 	  <= bit_d_9;

		if(last_is_control)
		begin
			parity_rec_d_gen <= !(bit_d_8^control[0]^control[1]);
		end
		else if(last_is_data)
		begin
			parity_rec_d_gen <= !(bit_d_8^data[7]^data[6]^data[5]^data[4]^data[3]^data[2]^data[1]^data[0]);
		end
	end
end

martinwhitaker, 2017-11-16:
Check there are no other assignments to control_r or parity_rec_c in your code, also that control_r is really 4 bits wide.

Other than that, we'll need a complete test case to investigate this. But it seems likely it's a bug elsewhere in your code, as your second, very similar, code fragment works correctly.

red0bear, 2017-11-16:
yes , on zero delay works well.  I just asked to be sure if it make right thing. During schedule process it change data while another one similar don't. Well if you don't see any error could i close this .

closed 2017-11-16 by red0bear