steveicarus/iverilog #175: SystemVerilog String support

svenka3, 2017-11-27:
Basic sting declaration works with -g2012

However 2 issues found recently :

1. parameter string MYSTR = "aa";

Errors out

2. String array method size() not supported (will raise a new issue with test)

Thanks 
Srini

steveicarus, 2020-12-20:
string.size() is now supported. I'm taking on the support for string parameters.

caryr, 2020-12-20:
Are you also planning to implement the rest of the string methods in both a constant and variable context?

steveicarus, 2020-12-20:
> Are you also planning to implement the rest of the string methods in both a constant and variable context?

Maybe. I'll see what comes up. At the very least, this is going to involve a rework of the parse.y rules for parameter types to use the data_type_or_implicit so will provide the infrastructure for a lot of stuff in the long run.

caryr, 2020-12-20:
Great. Keep in mind this can eventually supports arrays. I don't remember any other PRs. Would likely need to check all the ones here and on SourceForge.

Should we clone all the SourceForge items over here?

steveicarus, 2020-12-20:
We should completely deprecate the SourceForce bug reports database and
clone over to github anything we think is worth preserving.

caryr, 2020-12-27:
FYI I just pushed a patch to parse.y that allows parameters that use the fixed width integer data types to support using signed/unsigned in parameters (e.g. int unsigned is a 32 bit 2-state unsigned value).

steveicarus, 2020-12-27:
I've been busy with parse.y WRT parameter parsing, so be careful with edits
in that area. I'm working on a fairly long and involved task, so I expect
to have to do some conflict resolution, but try to not make my work harder
there.

closed 2020-12-28 by steveicarus