steveicarus/iverilog #178: Migrate to CMake?

docbrown, 2017-12-10:
Would the Icarus Verilog developers be open to migrating from Autotools to CMake? That would enable a much smoother developer experience on Windows with Visual C++ support, not to mention faster builds with Ninja and better IDE integration across all platforms. I'm happy to start writing the CMake build files, but I wanted to make sure a change like that would be welcome first.

martinwhitaker, 2017-12-14:
This is one for Steve to answer. I don't have strong views either way.

steveicarus, 2017-12-14:
Years ago, someone offered to do this, and I decided against it because
cmake wasn't as stable as it is now, and given the wide porting base, the
autoconf system seemed the better bet.

But cmake is much more mature then it was, and also Icarus Verilog
source itself is not the moving target it once was. So I'm OK with you
trying to make a cmake build environment.

The must-haves, obviously, are support for Linux, MacOS, and Windows.

Probably would be a good idea to keep the iverilog-devel mailing list
apprised of your progress, as well.

caryr, 2017-12-18:
I am basically with Steve on this one. The only downside I see is none of the core developers have any experience with cmake, while for better or worse we do have experience with the current autoconfig flow. Though this is not a good enough reason not to migrate if there are clear benefits. Could you provide some pros/cons for cmake versus autotools? Last time I tried to build something from source with cmake (a number of years ago) it was challenging to hack to get working in my environment which was fairly mainstream. For example does cmake solve the cross compile issues we have where some parts of the code are generated from compiled code that needs to be compiled with the host compiler not the cross compiler.

docbrown, 2017-12-18:
To me, the biggest advantage of CMake is that it allows developers to work in whatever environment they are most comfortable with. I happen to prefer using an IDE with autocomplete and refactoring tools, so CMake lets me use Visual Studio on Windows and CLion or VSCode on Linux. Mac developers can use XCode, if they wish. Or if you just want to use a more traditional command line environment, CMake will generate files for Make or Ninja.

Besides that, CMake definitions are overall simpler than those in `Makefile.in`. After a glance through the Makefiles in IVerilog, there is `a lot` of duplication. Most of the `Makefile.in` files can be replaced with 10-20 line `CMakeLists.txt` files. File installation can be done with simple `install()` calls. `cppcheck` can be handled at the top level. All of the `sed` transformations can be done by `string(REGEX)` or `configure_file`. The application of Windows-specific compile flags can be handled automatically with target interfaces. And so on.

CMake's main con is its syntax, but it isn't that bad once you get used to it, especially if you've used a language like Tcl before. And the documentation is pretty good with lots of examples.

You technically can build for the both the host and target systems with CMake, but you have to generate a second build tree for the host programs. The whole process can be automated, but it isn't ideal. Fortunately, it looks like all of the host programs (`version.exe`, `dosify`, and `draw_tt.exe`; let me know if I missed any) can be replaced with CMake scripts, so that shouldn't be a problem anyway.

steveicarus, 2017-12-18:
I think the right way to handle this is for you to go ahead and do your
work,
possibly in a branch, and present the results in the iverilog-devel mailing
list.

Most developers these days don't touch the makefiles anymore, so I really
doubt you'll get much push-back, so long as you "own it" and support your
changes for long enough for it to sink into the culture of the developers.

caryr, 2017-12-19:
The only thing I will add is when you present this it would be nice to have a simple overview regarding the things we are still likely to do (e.g. add a new file for compilation, add a new C function requirement, etc.).

closed 2022-02-26 by steveicarus