steveicarus/iverilog #179: Bug in VVP grammar? .port_info statements don't end in a semicolon.

phire, 2017-12-14:
I discovered this while writing my own toy parser for vvp files.

Every single statement must end with a semicolon. with the small exception of `.port_info` statements, which a defined in `vvp/parse.y` (line 771) as not ending in a semicolon. I'm guessing this was an issue with the initial implementation of the port info feature. I've checked and there are no other exceptions.

Likewise, tgt-vvp skips the semicolon when outputting `.port_info` statements. 

It looks like it's reasonably easy to fix, but there is a question of backwards/forwards compatibility. My quick testing shows that adding a semicolon to the end of the `.port_info` statement will parse in current versions of vvp (I suspect it's using the empty statement rule). This means we can just update tgt-vvp to include the semicolon.    
But if we update the grammar in `vvp/parse.y` to enforce semicolons then old vvp scripts will no-longer work in newer versions. It might be a good idea to make the semicolon optional, and maybe print a warning message. Or we can just add a comment to `vvp/parse.y` explaining why it's missing for future reference.  

I'm more than happy to submit a pull request. I just wanted to consult on the correct way of dealing with backwards compatibility. 

Thanks.

steveicarus, 2017-12-14:
Backwards compatibility within a release branch is sacred, and I would not
want
to see a change on this within the v10 branch.

Within the devel branch, if it can be done in a non-intrusive way (i.e.
allow vvp
to read older .vvp files) I'd be reasonably OK with that.

The .port_info statement has gone though iterations through the years, so
there
are all sorts of possibilities as to how we got here.

phire, 2017-12-16:
Looks like the easiest way to fix this non obtrusively is to do it in two parts:

 1. During the 11.0 dev cycle, change tgt-vpp to start outputting semicolons after `.port_info` statements.  
   This doesn't cause any problems because vvp currently accepts `.port_info` both with and without a semicolon.

 2. During the 12.0 dev cycle, change vvp to reject any `.port_info` statements without a semicolon as a syntax error.  
This avoids any backwards compatibility issues because all 12.0 files will already have semicolons, and vvp refuses to run files from older versions of vpp anyway.

Could even do part 2 as an `#ifdef` that automatically starts enforcing the stricter syntax as soon as the version number gets bumped to 12.0

caryr, 2017-12-17:
I believe you can just add this to V11 (devel) without all the special steps since it is not part of a release branch like V10. We often make changes in development that would not be allowed in the release branch. Of course Steve should confirm this matches his intentions.

PR #181 (2017-12-18): vvp grammar: .port_info statements should require/generate semicolons.
This PR attempts to fix #179 

There are two parts. The first commit simply fixes tgt-vvp so it includes a semicolon after all .port_info statements. This fix is inherently backwards compatible, as current versions of vvp accept the extra semicolon without issue (they think it's an empty statement).

The second commit fixes the parser to be more strict and syntax error when the semicolon is missing. Because this would cause backwards compatibility issues (any vvp files generated on 11.0 dev before this commit is merged will [... cut]
closed 2017-12-18 by steveicarus