steveicarus/iverilog #180: Question regarding vvp parameter option

feddischson, 2017-12-16:
Is there a way to pass data via vpp commandline option into the design?

A nice feautre would be to have a `-g` option which allows to set or to override a parameter.
With such an option, a compiled design could be run with different parameters.

martinwhitaker, 2017-12-16:
The -P option allows you to override a top level module parameter. The -D option allows you to define a macro. See the man page or the Wiki for details.

feddischson, 2017-12-16:
But both only exist for **`iverilog`** and not for **`vpp`** right?

martinwhitaker, 2017-12-16:
Yes, but you asked about setting or overriding parameters. Parameters are constants, evaluated at compile time. You can't possibly change them at run time.

If you want to pass values in at run time, use the $test$plusargs and $value$plusargs system functions. See the vvp man page for details.

caryr, 2017-12-18:
A specparam is the other choice, but we do not currently have support for updating them from a SDF file since they are basically treated as parameter in the compiler.

caryr, 2017-12-19:
Can this be closed or can you provide more details regarding exactly what you are trying to do?

feddischson, 2017-12-19:
@martinwhitaker and @caryr thanks for your help. The `$test$plusargs` is a solution for the moment.

I was trying to support Icarus for the  [vunit test framework](https://github.com/VUnit/vunit). The branch can be found [here](https://github.com/feddischson/vunit/tree/icarus_support) (see also [here](https://github.com/VUnit/vunit/issues/188)).

One point was the passing of test-information during runtime into the test-bench. All other simulators (as far as I can see) do this via a commandline flag when calling the simulator. Modelsim/vsim for example has as a `-g` option. 
This allows that the test and design can be compiled once and executed in parallel for different test-cases.

martinwhitaker, 2017-12-19:
plusargs is the method specified in the Verilog LRM, so is (...or should be) available in all simulators.

What capability does the ModelSim -g option provide that plusargs does not?

kraigher, 2017-12-20:
@martinwhitaker The use case for overriding parameters at elaboration time is to run the same test bench with different parameters passed to the DUT (Device Under Test). For example test a FIFO with different depths, data_widths, almost full settings etc. Modelsim, Riviera-PRO, Cadence Incisive, GHDL and Vivado XSIM all support such a feature for both VHDL and Verilog. I do not believe a parameter on the DUT instantiated in the test bench can be set via plusargs. How do you suggest to solve this use case with iverilog? Would you have to compile different versions of the test bench as separate output artifacts? They must be able to be run in parallel as well so "compiling over" the same artifact before simulation serially is not ideal.

@feddischson At least for runner_cfg string plusargs is probably ok to use. In Verilog version of VUnit the fact that the runner_cfg is a parameter is private implementation detail. This at least covers the core functionality, however VUnit also has features for running a testbench with different parameter values and that feature would be harder to implement.

martinwhitaker, 2017-12-20:
@kraigher, yes, I know that. As I said in my first reply to @feddischson, if you want to override a parameter, use the iverilog -P option. This is the equivalent of the NC-Verilog -defparam option. iverilog doesn't support separate compilation, so you always have to recompile if you want to re-elaborate. But the compilation stage only accounts for a small percentage of the overall compile/elaborate/codegen time, so that's not a big problem.

I don't see any way it's possible for any simulator to support overriding parameters **after** the design has been elaborated. Can you point me to an example?

kraigher, 2017-12-20:
@martinwhitaker In the case of the simulators I mentioned they override the parameter/generic before elaboration not after. Override after elaboration would just as you say not be reasonable. Basically there are three stages; compilation, elaboration and simulation. Compilation is mapped to one command such as 'vlog' and elaboration+simulation is mapped to another command such as 'vsim'. Forgive my ignorance but maybe icarus maps compilation+elaboration to one command and simulation to another? In that case how is it possible to support compiling one module without default values on its parameters that is then later instantiated several times with different parameter values?

martinwhitaker, 2017-12-20:
@kraigher, yes the Icarus 'iverilog' command is a combined compile+elaborate command, and 'vvp' is the run-time interpreter. As I said, there is no support for separate compilation. If you want to elaborate your design with different values on the parameters, you simply rerun the iverilog command, specifying the values with the -P option.

kraigher, 2017-12-20:
@feddischon Based on the answer from @martinwithaker this means that for iverilog the compile step must be included in every individual simulation. It also means that the normal compile step in VUnit is unnused when iverilog is the backend. Incremental compilation is meaningless for iverilog. At least I guess preprocessing could be done as a common compile step instead of for each simulation if so desired.

kraigher, 2017-12-20:
@martinwhitaker Is there a technical or other reason that you do not support separate compilation? I can see it being a problem in large projects where re-compile times are not neglible and can even dominate if the test bench sequence is short. With separate compilation it is possible to have incremental compilation to mitigate the compilation time from being a problem in the edit/compile/run development loop.

martinwhitaker, 2017-12-20:
@kraigher, I did consider trying to add support for separate compilation a while ago, but decided it wasn't worth the effort for two reasons:
1. iverilog runs through three main phases: parsing, elaboration, and code generatoion. Code generation is implemented in a separate DLL, and several different code generators are supported (just one of which generates code for the vvp interpreter). The bulk of the time is spent in elaboration and code generation - parsing is a relatively quick job. To make separate compilation worthwhile, more work would have to be done before elaboration, but that would require a fundamental redesign of the compiler architecture.
2. The iverilog compiler is quite fast, whereas the vvp interpreter is slow (circa. two orders of magnitude slower than simulators that compile down to native code). So even for short unit tests, run time usually dominates.

caryr, 2017-12-21:
If you understand that Icarus will likely never be used with large projects and for small to medium sized projects it is usually fast enough as is there is not a lot of incentive for us to work on a major project like restructuring the compiler (compile/elaboration stages). I believe our users would rather have us use what time we have available to work on adding support for missing features. There are also a few major rework items that effect functionality that we should work on first so I don't see restructuring Icarus to follow the parse, elaborate, run pattern happening anytime soon. If there are missing features you need to support vunit please let us know. I would personally like to see Icarus usable in a TDD flow.

kraigher, 2017-12-21:
@caryr @martinwhitaker Thank you for your elaborate answers. From them I conclude that iverilog requires a different flow than other simulators that VUnit supports. This should not inhibit VUnit supporting it but rather just make it slightly more difficult.

feddischson, 2017-12-23:
@kraigher @martinwhitaker Thank you for all your information, I was not aware of these details.
I will modify the [icarus support branch](https://github.com/feddischson/vunit/tree/icarus_support) that the compilation is done in the simulation step. 

@caryr The missing features which I discovered are listed [here](https://github.com/feddischson/vunit/blob/icarus_support/docs/missing_icarus/missing_icarus_features.rst)

feddischson, 2017-12-23:
@martinwhitaker 

Considering 
```
module parameter_example;
parameter my_param = "";
initial begin
  if( my_param == "" ) begin
    $error("my_param is empty");
  end else begin
    $display("my_param is %s", my_param);
  end
end
endmodule
```

How is it required to call `iverilog` to overwrite `my_param`?

`iverilog -P parameter_example.my_param="xyz" -o parameter_example parameter_example.sv -g2012 ` gives me the following error message:
`<command line>: error: invalid value specified for defparam: parameter_example.my_param`

martinwhitaker, 2017-12-23:
My guess is your shell is eating the double quotes. Try
```
iverilog -P parameter_example.my_param=\"xyz\" -o parameter_example parameter_example.sv -g2012
```
to prevent that.

caryr, 2017-12-24:
Regarding missing features you really should be using the devel branch to check this. Using development sounds scary for most people, but in Icarus land it should not be. In general stable versus development involves stability issues that most users do not care about. Other than short transient issues development should be stable in the sense that it functions correctly. Occasionally we submit something that needs to be quickly fixed, but the intention is that both development and stable always pass the regression tests and fundamentally development should pass more of the tests and stable gets marked as expected fail or not implemented for this new functionality.

When I was using Icarus at my day job I only used development.

martinwhitaker, 2018-09-28:
I think all the questions here got answered satisfactorily, so closing this.

closed 2018-09-28 by martinwhitaker