steveicarus/iverilog #182: counter under clock cycle strange behavior

red0bear, 2017-12-31:
Well here am I again. After many tries here trying to see some mistake by me, but I'm convinced me there is something strange on simulation.  
counter_neg is under negedge_clk and should change his value only in end of edge, but it change without arrive in end of negedge_clk. If do you think what I'm talking may be wrong i need notice you the flag is_control is in same edge clock and same always where  counter_neg in and works right while counter_neg not.

For this strange thing happen i just changed register on my transceiver from sequential to combinational "tx_dout_e and tx_sout_e". This should not cause any troubles to receiver but it do this. 

I'm happy to see what i was trying to explain but ..... i don't know what exactly could be. Here a VCD file and ScreenShot from from what showing the problem. If is possible to help i could be very glad.

The problem happen in new version from git hub and last month update.  

Happy new year people.  
 
[module_tb.vcd.zip](https://github.com/steveicarus/iverilog/files/1595264/module_tb.vcd.zip)

[image]

caryr, 2018-01-01:
A verbal description is almost never enough for us to help solve the problem. We need a simplified Verilog example that demonstrates the problem. In my experience creating this example often exposes the mistake/misunderstanding.  Even commercial simulators require an example for bug report, but often there because of an NDA the whole design can be provided. That doesn't work as well for an open source product since no one is paying us to debug their circuits.

What I have found best is to take the whole design and start removing/replacing large portions of the circuit with idea pieces around the block am seeing problems. If the functionality changes that is telling you something so try to use that to figure out what the issue is. The trick is to keep enough of the circuit to demonstrates the problem without having to keep the whole design. Think of this as your contribution to us solving the problem (i.e. you provide a simple easy to understand example and we try to understand why that is causing Icarus to fail).

red0bear, 2018-01-01:
i dont think so ... i have been tested with another one "cadence/altera" and they don't change dramatically the counter_neg. i see no trivial problem to resolve that's why i just make a verbal description. 

You can start see the example right here : https://github.com/GLADICOS/SPACEWIRESYSTEMC
Just try start a connection.

martinwhitaker, 2018-10-06:
If there is a race in your code, it is very likely you will see different behaviour when you use another simulator. As Cary said, if you want us to investigate this, you need to provide a simple test case that demonstrates the problem.

red0bear, 2018-10-06:
@martinwhitaker @caryr ... i understand the problem is if this problem is real or not and how to reproduce it. But like i said before it is observed in project ..... i really don't realize how it could be done in another way. i don't wish guys you waste time, i just have been change code to avoid that. If you wish close this, its ok.

martinwhitaker, 2018-10-08:
OK, closing.

closed 2018-10-08 by martinwhitaker