steveicarus/iverilog #183: Variable declarations in the $root scope are not supported

guyfleeman, 2018-02-16:
I'm compiling with -g2012 from master.

A quick trace led me to this, pform.cc:2694

```
if (is_compilation_unit(lexical_scope)) {
    VLerror(li, "sorry: variable declarations in the $root scope are not yet supported.");
    return;
}
```

I'm guessing this isn't an issue easily tackled by a new iverilog dev? I wouldn't mind taking a look at an approach to this feature but I want to be realistic with my time/expectations for now. My skill set is definitely closer to "easy" than "moderate" right now.

Is this already within the scope of an active developer on the project? Where does this feature fit in the roadmap of priorities? I checked the wikia and didn't see anything listed pertaining to this feature. The man page lists 2005 as the default standard, so I'm guessing 2012 is still in active dev?

I can provide a source file if you'd like. 

Thanks for all the hard work you put into this project.

martinwhitaker, 2018-02-17:
This should of course be variable declarations in the $unit scope, not the $root scope.

I did the ground work for supporting this a few months back, and it should be fairly easy to complete. Just removing that old error check may be enough to get most things working. But when I started writing some regression tests, I exposed a whole load more bugs in the SystemVerilog support, and I lost the will to go on...

I'll try to push this a little further forward.

martinwhitaker, 2018-02-18:
This restriction is now removed, along with the similar one for parameters/localparams. There's bound to be bugs still to find, but basic stuff should work.

closed 2018-02-18 by martinwhitaker