steveicarus/iverilog #184: VPI -- cbAtStartOfSimTime cannot be registered multiple times for the same sim-time

yanghao, 2018-03-01:
For example:
registering a callback of reason cbAtStartOfSimTime is causing a segmentation fault. If setting vpi_time to be "0" then no issue, callback registered and called properly.

cb_data.vpi_time are the same value (e.g. at time 1), then vpi_register_cb(&cb_data) is called multiple times (in initial statement at time 0), second time registration causing a segmentation fault with below backtrace:

```
Registering 1 callback ...
Registering 2 callback ...

Thread 1 "vvp" received signal SIGSEGV, Segmentation fault.
0x00005555555fa627 in schedule_event_ (cur=0x555555889b30 <generic_event_heap+80>, delay=<optimized out>, 
    select_queue=select_queue@entry=SEQ_START) at schedule.cc:657
657			  cur->next = ctim->active->next;
(gdb) bt
#0  0x00005555555fa627 in schedule_event_ (cur=0x555555889b30 <generic_event_heap+80>, delay=<optimized out>, 
    select_queue=select_queue@entry=SEQ_START) at schedule.cc:657
#1  0x00005555555fbeec in schedule_at_start_of_simtime (obj=obj@entry=0x555555b0e350, delay=<optimized out>) at schedule.cc:940
#2  0x0000555555623863 in make_afterdelay (data=<optimized out>, simtime_flag=simtime_flag@entry=true) at vpi_callback.cc:392
#3  0x0000555555623d92 in vpi_register_cb (data=data@entry=0x7ffff664e1f0 <d+112>) at vpi_callback.cc:577
#4  0x00007ffff644c725 in usw_callback_time (i=i@entry=2, time=time@entry=10, callback=callback@entry=0x7ffff644c970 <test2_cb>)
    at usw_vpi.c:116
#5  0x00007ffff644cd6e in hello_calltf (user_data=<optimized out>) at test.c:71
#6  0x000055555563393a in vpip_execute_vpi_call (thr=thr@entry=0x555555aaef80, ref=0x555555add640) at vpi_tasks.cc:941
#7  0x00005555555f0511 in of_VPI_CALL (thr=0x555555aaef80, cp=<optimized out>) at vthread.cc:5674
#8  0x00005555555e8242 in vthread_run (thr=0x555555aaef80) at vthread.cc:665
#9  0x00005555555fc0ef in schedule_simulate () at schedule.cc:1130
#10 0x00005555555b7008 in main (argc=<optimized out>, argv=0x7fffffffde18) at main.cc:501
(gdb)
```
And the source code at line of 657:

```
653       case SEQ_START:
 654         if (ctim->start == 0) {
 655           ctim->start = cur;
 656         } else {
 657           cur->next = ctim->active->next;
 658           ctim->active->next = cur;
 659           ctim->active = cur;
 660         }
 661         break;
 662 
```
Which comes from patch:

commit 1b7a42ffcd6d54bc3dee2ad9c717656cbed7e586
Author: Stephen Williams <steve@icarus.com>
Date:   Sun Nov 1 09:26:09 2009 -0800

    Implement cbAtStartOfSimTime callback.

OS: ubuntu 17.10
icarus: v10 branch latest version

Best Regards,
Yanghao

martinwhitaker, 2018-03-09:
I've not managed to reproduce the fault from your description. Please could you provide a small example that fails in this way (both the Verilog and the accompanying 'C' code).

yanghao, 2018-03-09:
```
git clone https://github.com/yanghao/iverilog-vpi-failure
cd iverilog-vpi-failure
run
```

martinwhitaker, 2018-03-10:
OK, I confirm this failure in the v10 branch, but it appears to be fixed in the master branch:
```
% run
VCD info: dumpfile dff.vcd opened for output.
Hello, World!
Registering 1 callback ...
Registering 2 callback ...
Registering 3 callback ...
Registering 4 callback ...
Registering 5 callback ...
Registering 6 callback ...
Registering 7 callback ...
Registering 8 callback ...
Registering 9 callback ...
Hello from test2_cb
Hello from test2_cb
Hello from test2_cb
Hello from test2_cb
Hello from test2_cb
Hello from test2_cb
Hello from test2_cb
Hello from test2_cb
Hello from test2_cb
abc...
```
I'll try to find the change that fixed it.

martinwhitaker, 2018-03-10:
I've back-ported the fix to the v10 branch.

closed 2018-03-10 by martinwhitaker