steveicarus/iverilog #187: Variables in command files not support in +define+

jimmysitu, 2018-04-07:
Hi, All
I just try the -c option for command file for iverilog, it seems +define+ is not support $(varname), ${varname}, here is my code and cmd file
test.v
```verilog
module print;
    always@(*)begin
        if (`PRINT_COND) begin
            $display("Printing...");
        end
    end
endmodule

module test;
    wire print_cond = 1'b1; 
    print dut();
endmodule
```
ivlcmd
```
+define+PRINT_COND=$(TB).print_cond
```
run in bash
```bash
export TB=test;iverilog test.v -civlcmd
```
Error
```
test.v:4: error: malformed statement
test.v:3: error: Malformed conditional expression.
test.v:6: syntax error
test.v:9: error: invalid module item.
```
Any idea?

martinwhitaker, 2018-04-07:
Yes, variable substitution is not supported in the +define+ value. I would guess Steve did that deliberately to avoid the complication of supporting escape characters.

To do what you want, write yourself a shell script that calls iverilog and uses the '-D' option to predefine macros. Then the shell will do the substitution.

steveicarus, 2018-04-07:
Variable substitutions do work, but only for file and directory names.

   http://iverilog.wikia.com/wiki/Command_File_Format

You can include sub-command files to carry configuration specific
definitions,
and use variable substitution to select sub-command files.

martinwhitaker, 2018-04-09:
Variable substitution is also performed on timescales, parameter overrides, integer width values,... in fact as far as I can see, all values except +define+ values.

martinwhitaker, 2018-09-28:
I think this was reopened by accident...

closed 2018-09-28 by martinwhitaker