steveicarus/iverilog #188: error: Unable to elaborate r-value: glbl.GSR

kdpatino, 2018-04-19:
Hello!

I was trying to simulate with the unisims library from Xilinx and I got:
`error: Unable to elaborate r-value: glbl.GSR`

It means that the glbl.v file didn't set as a global variable. Someone know how I will add this file to set up as a global configuration
```
$(IVERILOG) -g2012 -gno-io-range-error -o simulation/$(DESIGN)_TB.vvp $(VINCDIR) $(SRC) $(SIM_SRC) -s $(DESIGN)_TB -y $(ISE_SIM_ROOT) -y $(ISE_SIM_ROOT)/unisims -y $(ISE_SIM_ROOT)/unimacro -y $(ISE_SIM_ROOT)/simprims
```
Thank You in advance

Regards,
Kevin Patino

martinwhitaker, 2018-04-22:
Just add $(ISE_SIM_ROOT)/glbl.v to the command line.

martinwhitaker, 2018-09-28:
No response, so I'm assuming this is now resolved.

closed 2018-09-28 by martinwhitaker

iDoka, 2019-01-10:
@martinwhitaker if `glbl.v` absent in compilation stage we giving different kind of issue (BTW, @kdpatino already included `glbl.v` in command line as `-y $(ISE_SIM_ROOT)`).

#188 should be open. Still no way to resolve issue.

iDoka, 2019-01-10:
**update:**
@lomax help me to solve this issue: the root of cause that glbl used, but not instantiated into netlist.
What we should to do to resolve issue:

1. add `$(ISE_SIM_ROOT)/glbl.v` to command line (approach with adding `-y $(ISE_SIM_ROOT)` doesnt work in case of uninstantiated glbl)
2. add **additional** top to command line: `-s glbl` (i.e. total: `-s $(DESIGN)_TB`)
3. That's all!

martinwhitaker, 2019-01-10:
Alternatively, don't use the '-s' option at all. Then the compiler will automatically add all uninstantiated modules to the top-level module list.