steveicarus/iverilog #191: SystemVerilog loop-controll support (break, continue)

feddischson, 2018-05-13:
Hi,

it would be greate to have SystemVerilog's loop-control statements suporrted by iverilog.

The following gives the feedback `break_example.sv:8: sorry: break statements not supported.`:
```systemverilog
module break_example;
integer i;
initial begin
  for( integer i=0; i < 5; i=i+1 ) begin
    $display( "%d\n", i);
    if (i==2) begin
      break;
    end
  end
end
endmodule
```

This example responds `break_example.sv:8: error: malformed statement`:
```systemverilog
module continue_example;
integer i;
initial begin
  integer i;
  for( i=0; i < 5; i=i+1 ) begin
    $display( "%d\n", i);
    if (i==2) begin
      continue;
    end
  end
end
endmodule
```

Icarus Verilog version 11.0 (devel) (s20150603-536-gca013857)

LarsAsplund, 2021-09-27:
Would be great to have this and #192. All that is required for VUnit support.

caryr, 2021-09-28:
Adding this should not be too complicated since in VVP it is just unconditional jumps to the appropriate loop labels. We would also want the vlog95 generator to correctly generate code so the loops may need to know they have these constructs to generate the correct code. In VVP we could likely just always generate this code/extra labels, but this would lead to extra unneeded code in the vlog95 output. Which could create confusion when looking at the generated code.

thirtythreeforty, 2021-12-31:
I have implemented a nearly-working demo of iverilog support for VUnit/vunit#188; agree that #191 and #192 are the only language features it needs. Would be awesome!

balanx, 2022-08-18:
hope to support (break and continue)

closed 2023-01-16 by steveicarus