steveicarus/iverilog #193: SystemVerilog immediate assertion statement

feddischson, 2018-05-13:
Hi,

it would be greate to have SystemVerilog's immediate assertion statements working in iverilog.
The following example respondes `assertion_example.sv:5: sorry: Simple immediate assertion statements not implemented.`

```systemverilog
module assertion_example;
integer i;
initial begin
  integer i;
  assert( 1==1 ) else
  begin
    $display( "something went wrong");
  end
end
endmodule
```

jopdorp, 2018-12-25:
This would be great for testing!

byllgrim, 2019-09-17:
Would this be much work to implement?

Asserts seem superficially similar to if-statements, so maybe one can learn from how if-statements are implemented and it ain't so hard?

I don't have loads of free time on my hands, but it would be nice to document some clues in this thread and maybe I or someone else looks at it?

martinwhitaker, 2019-09-19:
It shouldn't be much work. The hard parts are already done - supporting it in the parser and implementing the $error system call. For the rest, it's really just an if test with the true and false clauses swapped, so we can use existing code in the compiler for that.

I'll look at implementing this soon.

martinwhitaker, 2019-09-20:
Procedural immediate assertion statements are now supported in the master branch. $assertcontrol is not supported, but you can globally disable assertions at compile time using the -gno-assertions compiler option.

assert and assume statements should work as expected. cover statements are parsed but otherwise ignored.

closed 2019-09-20 by martinwhitaker

jopdorp, 2019-11-07:
@martinwhitaker 
I tried this on the master branch.
I compiled it from mingw64

$ iverilog  -gno-assertions -o example.vvp not_n2t.sv not_n2t_tb.sv

not_n2t_tb.sv:10: error: Enable of unknown task ``assert''.
not_n2t_tb.sv:12: error: Enable of unknown task ``assert''.
2 error(s) during elaboration.

this is the module code:

module not_n2t_tb();
    reg in;
    reg expected;
    wire out;

    not_n2t gate1(in, out);

    initial begin
        in = 1'b0;
        #1 assert (out == ~in);
        #1 in = 1'b1;
        #1 assert (out == ~in);
    end
endmodule

martinwhitaker, 2019-11-07:
@jopdorp, assertions are a SystemVerilog feature. Try adding -g2012 to the compiler command line.

jopdorp, 2019-11-08:
that worked! thanks so much for the tip! I didn't know that there was a commandline argument that has to be passed for enabling the supported SystemVerilog features.

This is great for my repo https://github.com/jopdorp/nand2tetris-verilog
I'm porting the nant2tetris chip design course to verilog and have been using system verilog assert statements for the tests.
Until now I've been using the model sim simulator, but from the stat I've wanted to use icarus iverilog, because it's open source, and that's awesome!

I'll be u[dating the readme :D

Johnlon, 2020-03-21:
I have checked out the git repo and built it and running with -g2012 but still getting 
"Simple immediate assertion statements not implemented."
Any clues?

martinwhitaker, 2020-03-21:
@Johnlon,
1. Make sure you are building from the master branch, not the v10 branch.
2. Make sure you are running the version of the compiler you built, not an older version. 'iverilog -v' will report exactly what is being run.

mcandre, 2021-09-28:
Which git commit hash has this fix?

Which release tag has this fix?

caryr, 2021-09-28:
If you want the most functionality you really want to be using the development branch. Because of the test suite we have it is usually stable and has all the latest functionality.

mcandre, 2021-09-28:
Yes, but in the documentation for my applications I would like to be able to tell my contributors, grab _X_ version of iverilog in order to work on this project. For now, all I could hand them is a generic git hash.

Is there a milestone, a list of specific issues to resolve before the next release version will be tagged?

Johnlon, 2021-09-29:
I record the latest git hash that I've seen work on my project.

toolness, 2023-12-17:
FWIW, this feature seems to be in version 12.0 stable as of today!

However, I can't seem to arrive at a situation where iverilog understand the assertions _and_ vvp runs the assertions along with dumping a VCD file.  When I pass `-g2012` into iverilog, it works fine, but when I run vvp, I get this:

```
VCD warning: $dumpvars: Package ($unit) is not dumpable with VCD.
```

This is a bummer, since ideally I'd like both; if the assertions fail, I want to be able to inspect the VCD using GTKWave to debug and figure out what's amiss.

martinwhitaker, 2023-12-17:
@toolness , that's only a warning. It will still output a VCD file, and you will be able to inspect the contents of modules. If you need to inspect the contents of packages as well, use the FST file format instead of VCD.

toolness, 2023-12-21:
Ah, you're right, thanks!

Unfortunately I am too much of a noob to know what the difference between a package and a module is at the moment, but it is indeed outputting the VCD file despite the warning, with all the signals I'm interested in.

FWIW, I did also briefly try using FST (which required passing `-fst` to `vvp`) but I couldn't figure out how to make GTKWave open it will all the signals in view (the advice [here](https://sourceforge.net/p/gtkwave/mailman/message/35204673/) seems to only work for VCD files), so I guess I'll just stick with VCD for now.