steveicarus/iverilog #194: SystemVerilog String indexing

feddischson, 2018-05-13:
Hi, 

I discovered, that string-indexing fails during elaboration:

```systemverilog
module string_example;
  function int example( string my_string );
    if( my_string[0] ) begin
      return 1;
    end else begin
      return 0;
    end
  endfunction
endmodule
```

The example above results in `string_example.sv:3: error: can not select part of scalar: my_string`

Icarus Verilog version 11.0 (devel) (s20150603-536-gca013857)

steveicarus, 2019-10-29:
Icarus Verilog does support string indexing, but this example triggers a bug where it is not handled within function definitions. Specifically, indexing the string argument is what is failing here.

steveicarus, 2019-10-30:
Fix pushed to git master.

closed 2019-10-30 by steveicarus