steveicarus/iverilog #197: VPI: load tracing with vpiDriver and vpiLoad

yanghao, 2018-05-25:
It seems iverilog does not yet support load tracing in its vpi implementation (no vpiDriver/vpiLocalDriver/vpiLoad/vpiLocalLoad defined).

Are there an alternative way to do this in iverilog or is this feature planned to be supported in the future?

Best Regards,
Yanghao

martinwhitaker, 2018-05-31:
Apart from the $countdrivers system function (which is implemented), I don't know of any other way to get such information.

When I have some free time, I'll look at how hard it would be to add these functions.

yanghao, 2018-07-08:
Thanks @martinwhitaker 

The use case I have is to automatically detect reg/wire being driven already through VPI, because in this case assigning new values to an already-driven nets through vpi_put_value() will not have any effect. So better already warns user loudly at this stage.

I implemented the load tracing code through VPI initially for icarus only to find it is not supported, and tried same thing on vcs it works properly.

I did not find this $countdrivers() C counterpart from any of the iverilog header files, so I guess it is only available from within verilog code, which I cannot use due to all of my stuff is supposed to go through VPI only.