steveicarus/iverilog #198: $displayo of $signed leads to internal error (vpi_get_value for format=2 not implemented)

quark17, 2018-05-31:
For the attached example, I get this output:
```
> iverilog test.v
> ./a.out
r1        =  -9238
in binary = 1101101111101010
internal error: vpi_get_value(<format=2>) not implemented for __vpiVThrVec4Stack.
vvp: vpi_vthr_vector.cc:373: virtual void __vpiVThrVec4Stack::vpi_get_value(p_vpi_value): Assertion `0' failed.
Abort
```

If I comment out the $signed() call in $displayo(), there is no internal error.

ahogen, 2018-05-31:
The assert you're hitting is here:

ahogen, 2018-05-31:
So it's because the Octal case hasn't been implemented

Octal is format num 2:
https://github.com/steveicarus/iverilog/blob/c706c5dd90906ccf44db4e77cfe36901eb39123f/vpi_user.h#L241

And `vpiOctStrVal` isn't handled in the [aforementioned case statement](https://github.com/steveicarus/iverilog/blob/6e49ab10ec13a7c90deb07195724e917eb5188ec/vvp/vpi_vthr_vector.cc#L345-L378).

martinwhitaker, 2018-06-12:
I've pushed a fix for this to both the master and v10 branches. Feel free to reopen this issue if you find any problems with it.

closed 2018-06-12 by martinwhitaker