steveicarus/iverilog #199: divide overflow is not detected for verilog 32-bit integers.

oskarguan, 2018-06-11:
I'm using iverilog 10.1 on x64 win7 system with "iverlog -v" outputs:

$ iverilog -v
Icarus Verilog version 10.1 (stable) (v10_1_1)

Copyright 1998-2015 Stephen Williams

When I simulating my serial divider, I found that divide overflow of 32-bit integers:
$signed(32'h8000_0000) / $signed(32'hFFFF_FFFF) 
is not detected by vvp and will stall the program when evaluating the expression.

My test program is:

module divov1;
  reg [31:0] q1;
  reg [31:0] q2;
  reg [31:0] quotient;
  initial begin
    q1 = 32'h8000_0000;
    q2 = 32'hFFFF_FFFF;
    quotient = $signed(q1) / $signed(q2);
    $display("quotient = 0x%08h;", quotient);
    $finish;
  end
endmodule // divov1

Executing the test program never returns to the shell (cygwin) or cmd.exe.

$vvp divov.vvp

I also tested wider (64-bit) and narrower (16-bit) integers. Both of them didn't stall the program.

The 64-bit test program is:

module divov2;
  reg [63:0] q1;
  reg [63:0] q2;
  reg [63:0] quotient;
  initial begin
    q1 = 64'h8000_0000_0000_0000;
    q2 = 64'hFFFF_FFFF_FFFF_FFFF;
    quotient = $signed(q1) / $signed(q2);
    $display("quotient = 0x%016h;", quotient);
    $finish;
  end
endmodule // divov2

The output of 64-bit test program is:

$ vvp divov2.vvp
quotient = 0x8000000000000000;

The 16-bit test program is

module divov3;
  reg [15:0] q1;
  reg [15:0] q2;
  reg [15:0] quotient;
  initial begin
    q1 = 16'h8000;
    q2 = 16'hFFFF;
    quotient = $signed(q1) / $signed(q2);
    $display("quotient = 0x%04h;", quotient);
    $finish;
  end
endmodule // divov3

The output of 16-bit test program is:

$ vvp divov3.vvp
quotient = 0x8000;

martinwhitaker, 2018-06-12:
Running on x86_64 Linux, I get different results:
````
% vvp divov1.vvp
quotient = 0x80000000;
% vvp divov2.vvp
Floating exception (core dumped)
````
I suspect the difference is that a 'long int' is 64 bits in Linux, but 32 bits in Windows, and if we fix the fault in Linux, it will also cure the fault in Windows.

martinwhitaker, 2018-06-12:
I've pushed a fix for this to both the master and v10 branches. This fixes the problem in Linux (and similar problems when using the modulus operator and when using constant expressions. Please let us know if this also fixes the problem you are seeing in Windows.

oskarguan, 2018-06-14:
I built the master branch on MINGW64 and MSYS2 with uname -a output:

$ uname -a
MINGW64_NT-6.1 oskarguan-win7 2.10.0(0.325/5/3) 2018-02-09 15:25 x86_64 Msys

GCC and all dependent packages were installed with MSYS2's pacman. 

The upper 3 testcases terminated normally.

$ vvp divop1.vvp
quotient = 0x80000000;

$ vvp divop2.vvp
quotient = 0x8000000000000000;

$ vvp divop3.vvp
quotient = 0x8000;

This issue seems solved.

closed 2018-06-14 by oskarguan