steveicarus/iverilog #200: How hard would it be to add in basic support for SystemVerilog interfaces?

fl4shk, 2018-06-12:
(Note:  This is a copy/paste of my response to another issue)

I'm kind of interested in contributing to Icarus Verilog specifically on this issue. How hard would it be to add in support for an interface that simply consists of inputs, outputs, and parameters? Support for interfaces with only as much functionality as this would be really nice:

```
interface Example #(parameter N = 8)
  (input logic clk, rst_n, input logic [N_MINUS_1 : 0] in_data, 
  output logic [N_MINUS_1 : 0] out_data);
localparam N_MINUS_1 = N - 1;
endinterface
```

fl4shk, 2018-06-12:
Oh yes one other thing:  I mean for using these interfaces in module ports.

martinwhitaker, 2018-06-14:
The short answer is that it's not an easy task - but please don't let that put you off having a go!

The good news is that I have already added support for defining and instantiating interfaces. Internally they are implemented as a variant of modules, so all the functionality they have in common with modules should just work.

What's left to be done is enable their use as module ports. First you would need to modify the parser to recognise the syntax. This is likely to be harder than it should be because Icarus uses a bison-generated parser. The SystemVerilog grammar is very definitely not LALR, so you are likely to have to find workarounds for shift/reduce conflicts. The files to start looking at are parse.y and pform.cc. To help debug, you can set the pf_path variable in main.cc to cause the compiler to dump the parser output to a file.

Once you have the parser working, the code that handles module port connection is found in the function PGModule::elaborate_mod_ in elaborate.cc.

Don't hesitate to ask for advice.

fl4shk, 2018-06-17:
I do have some experience with Bison.  I used it briefly for a small project before I switched to ANTLR.  I've used ANTLR a number of times now for various, so I'm pretty used to it for defining the grammar.

The pointers towards what source files to edit and what edits to perform are really helpful.

I just did a small test with an interface and the Vlog95 target, and indeed it's as you said:  interfaces are implemented as modules.

Slightly off topic, but I have a use case for the Vlog95 target:  I have a device that I want to target that I have to use ISE for, which I can't use SystemVerilog with, hehe.

I'm expecting that interfaces in the vlog95 target will "just work" once everything is implemented due to them being implemented as variants of modules.

fl4shk, 2018-06-18:
I'm attempting to append to `port_declaration` as follows:
```
/* FL4SHK code:  attempt to support interfaces as module ports */
 | attribute_list_opt IDENTIFIER gate_instance 
	{
		perm_string type_name = lex_strings.make($2);
		// Need more stuff here
		delete[]$2;
	}

```
The "Need more stuff here" part is what I think I need help with.  How does one

steveicarus, 2018-06-18:
This conversation is best held on the iverilog-devel mailing list, so that
others
who are interested can follow along.

fl4shk, 2018-06-18:
Alright.  I'll join that.

closed 2018-06-18 by fl4shk