steveicarus/iverilog #201: ICARUS_VPI_CONST

drnmiller, 2018-06-27:
Using vpi_user.h with c++11 generates warnings as follows:

```
vpi.cpp:395:5: warning: ISO C++11 does not allow conversion from string literal to 'PLI_BYTE8 *' (aka 'char *') [-Wwritable-strings]
    VPI_REGISTER_FUNC(avr_create);
    ^
vpi.cpp:137:25: note: expanded from macro 'VPI_REGISTER_FUNC'
    tf_data.user_data = "$" #name;              \
                        ^
```

I see there was a discussion about const correctness in 2010 and the introduction of ICARUS_VPI_CONST but it is unclear how to use this correctly. Adding `-DICARUS_VPI_CONST=const` to `CXXFLAGS` doesn't work because it is redefined to nothing in vpi_user.h.

Is there a way to use ICARUS_VPI_CONST short of changing the define in vpi_user.h?

Perhaps change vpi_user.h to:
```
#ifndef ICARUS_VPI_CONST
#define ICARUS_VPI_CONST
#endif
```
so c++11 can compile without warnings?

martinwhitaker, 2018-09-28:
Sorry it's taken a while for anyone to respond to this. Your suggested change to vpi_user.h seems sensible to me. However, having tested it, defining `ICARUS_VPI_CONST=const` introduces a number of new warnings when building Icarus, so there's still a bit more work to be done.

martinwhitaker, 2018-09-30:
I've fixed vpi_user.h so you can override the definition and fixed the resulting warnings. Be aware that if you define ICARUS_VPI_CONST in your global CPP_FLAGS or CXX_FLAGS, that will propagate through to iverilog-vpi,.

The version of GCC I'm using doesn't produce the warning you report, so I don't know if this fixes your issue.

martinwhitaker, 2018-10-06:
I've added ICARUS_VPI_CONST=const to the compiler and runtime makefiles, so I believe this issue is now fixed.

closed 2018-10-06 by martinwhitaker