steveicarus/iverilog #202: Specify dumpfile via command line?

Wren6991, 2018-07-13:
I am migrating a project from a non-FOSS simulator to iverilog. I can run the testbenches and get my pass/fail output (yay!), but it seems like the _only_ way of specifying the dump file and scope is by adding lines to my test bench. There is no mention at all of dump file path in `man vvp`, and some searching finds a thread from 2009 with people discussing this missing feature (and some 10-year-out-of-date discussion of the necessary changes). It is especially odd that dump _format_ can be specified on the command line, but not dump _path_.

I have never seen a simulator that requires testbench code to be modified to get waveform dumps. In commercial tools, the division of labour is normally that your testbench provides stimulus and assertions, and the tool is your means of observing the internals of your design. Having to add lines to testbenches to support a tool change feels odd.

I also have a need to generate and compare multiple VCDs (or equivalent) from multiple runs of the same testbench, with some memories initialised with different hex files. This does not seem to fit into the existing iverilog flow.

- Am I wrong? I have spent a fair amount of time searching/poking around, but the ideal answer is that I'm an idiot and there is a simple option where you can specify dump path on the command line
- If not, how much code is required to add this? I'm more into RTL than software these days, but with a couple of pointers I could implement this

ahogen, 2018-07-14:
For what it's worth, I haven't seen this option either. I'd like it as well, so you're definitely not alone there.

steveicarus, 2018-07-14:
This shortcoming can be worked around using plusargs. Look into the
$value$plusargs
system function.

Wren6991, 2018-07-15:
Okay specifying dumpfile via plusargs is a great workaround for generating multiple VCDs, thanks.

I will keep on poking at Icarus source, at the very least for my local copy, so that I can revert my `tb_*.v` patch and keep my Verilog tool-agnostic.

In principle, would you be happy to accept a patch that added a cmdarg which had the same effect as adding a `$dumpfile`, and a `$dumpvars` scoped to the top-level module?

steveicarus, 2018-07-16:
The place to implement that would be in the implementation for the
$dumpvars system task.
It is implemented using VPI, and the VPI has access to plus-args.

marcio-ao, 2019-03-06:
Hi, I'm a verilog newbie and I came across this thread. I am trying to make a Makefile for running Verilog simulations. Here is what I have so far:

```
all: step_doubler_tb.lxt

step_doubler_tb.vvp: step_doubler_tb.v step_doubler.v

%_tb.vvp: %_tb.v %.v
	iverilog -o $@ $^

%_tb.lxt: %_tb.vvp
	vvp -l $@ $^ -lxt2
```

The problem is that when I run this, gtkwave complains that the generated "step_doubler_tb.lxt" is not a valid dumpfile. When I place a "$dumpfile("step_doubler_tb.lxt");" in the "step_doubler_tb.v" file and remove the "-l $@" from the Makefile, things work.

Can somebody explain how I can get the desired effect? I don't know what is meant by $value$plusargs.

martinwhitaker, 2019-03-06:
The -l option allows you to redirect standard output to a file, so is unrelated to value dump files.

Personally I wouldn't bother with $value$plusargs. Just add
````
initial begin
  $dumpfile(`DUMP_FILE_NAME);
  $dumpvars
end
````
to your testbench, then compile with (for example)
````
iverilog -D 'DUMP_FILE_NAME="mydumpfile.lx2"'
````
(this is from a shell script - the single quotes are there to ensure the double quotes are included in the macro value).

You only need to use $value$plusargs if you need to change the file name without recompiling.

marcio-ao, 2019-03-06:
@martinwhitaker: Thank you! This was exactly the hint I needed!

The Makefile rule looks like gobbledygook, but it does work nicely!

```
%_tb.vvp: %_tb.v %.v
	iverilog -D'DUMP_FILE_NAME="$(patsubst %_tb.vvp,%_tb.lxt,$@)"' -o $@ $^
```

closed 2023-02-26 by steveicarus