steveicarus/iverilog #203: Output LXT2 files not always flushed on exit

alexforencich, 2018-07-27:
I'm doing quite a bit of cosimulation with MyHDL and iverilog.  I am running in to issues with truncated or invalid LXT2 files.  This seems to be happening when the cosimulation ends early due to a failed assert in my python-based testbench and I presume MyHDL/Python kills iverilog with some sort of signal.  It's worth making that more robust so that gtkwave can more reliably open the dump file.

martinwhitaker, 2018-09-29:
vvp already caught the SIGINT signal (treating it as an implicit $stop). I've pushed a change to the master branch that makes it catch the SIGHUP and SIGTERM signals as well (treating them as an implicit $finish).

martinwhitaker, 2018-10-06:
I believe this issue is now fixed.

closed 2018-10-06 by martinwhitaker

alexforencich, 2019-01-05:
Yeah, this is definitely not fixed yet.  I'm not sure how python is shutting down icarus verilog, but I am still consistently getting broken lxt2 files when testbench assertions fail.

alexforencich, 2019-01-05:
I also occasionally get bad lxt2 files with normal simulation termination.  Sometimes adding a delay before ending the simulation in the testbench helps.  Perhaps there is a race condition of some sort here?

martinwhitaker, 2019-01-06:
The lxt2 close_dumpfile() function is passed to atexit(3), so should get called if the process terminates normally. You need to find out how python is terminating iverilog.