steveicarus/iverilog #204: What is the correct syntax for include files?

abdullahyildiz, 2018-08-27:
I have an include file (parameters.v) such as

```
`define DEPTH = 8
`define WIDTH = 198
```

I use these values in actual module such as

```
`include "parameters.v"

module foo(
input wire [`WIDTH-1:0] data,
input wire [`DEPTH-1:0] addr,
...
);
```

endmodule

When I run iverilog as

`iverilog -g2005 -I. sim/tb_foo.v foo.v`

it gives errors like

```
foo.v:12: syntax error
foo.v:1: Errors in port declarations.
foo.v:24: syntax error
foo.v:23: error: syntax error in parameter value assignment list.
foo.v:22: error: Invalid module instantiation
```

svenka3, 2018-08-27:
You define has extra =, please try:

```
`define DEPTH  8
`define WIDTH  198
```

abdullahyildiz, 2018-08-27:
Oh, sorry 👍 . I missed it.

closed 2018-08-27 by abdullahyildiz