steveicarus/iverilog #206: Documentation of what SystemVerilog features are supported

jh3141, 2018-09-07:
The current documentation is self-contradictory, suggesting that anywhere from almost no to a reasonable proportion of SystemVerilog features are supported.  All I can really ascertain from it is that support isn't complete.

Is there anywhere that the current status of SV support is documented?  If not, is there any chance of putting a brief summary up on the wiki?

tlf30, 2018-09-08:
I too would like to know as well.

martinwhitaker, 2018-09-08:
I agree this would be nice to have. Unless Steve can provide a list of what features he implemented, it needs a volunteer to do the legwork. The place to start would the test suite - if a feature is supported, there should be a regression test for it.

svenka3, 2019-02-06:
@martinwhitaker - where can I find the test-suite mentioned please?

Thanks
Srini

martinwhitaker, 2019-02-06:
@svenka3 - https://github.com/steveicarus/ivtest. More details can be found at https://iverilog.wikia.com/wiki/Developer_Guide#Testing_Your_Work

cmarqu, 2020-07-06:
This table has an "icarus" column: https://symbiflow.github.io/sv-tests/

martinwhitaker, 2020-07-07:
@cmarqu, it does, but it isn't very reliable. For instance it shows Icarus failing 52 tests in the ivtest suite. And there are lots of false negatives on other tests. Just picking one at random, https://symbiflow.github.io/sv-tests/#Icarus|5.7.1|number_test_0 fails because it is just a code fragment. Icarus expects a complete design.

caryr, 2020-07-07:
That table is slightly misleading since some of the actual test failures do not match what is being tested. It does point to a few things we could fix that would significantly increase the passing percentage. It's also interesting to see that Icarus scores low on it's own test suite so there are certainly some issues with that. It looks like the generation information is getting lost and they are running everything as -g2012. My guess is some of the original Verilog tests are having issues when run as SystemVerilog and this is likely impacting all columns.

I'll try to find some time to see if Icarus can be integrated better and triage things at a high level.