steveicarus/iverilog #208: c=1?$signed(a)>>>1:a>>1 appears miscompiled

tommythorn, 2018-09-23:
This is a dramatically reduced test case (the original looks nothing like this).

```
module t;
reg z=1;always#5z=!z;reg[31:0]a=0,b=0,c,d;
always@(*)begin
  c=1?$signed(a)>>>1:a>>1;
  d=$signed(a)>>>1;
end
always@(posedge z)begin{b,a}<={b,a}+'h1000;if(d!=c)$display("F");end
endmodule
```

In my understanding this should never print anything, but it does for me.

I'll try to rebuild from git, but Mac Brew is up to date.

tommythorn, 2018-09-23:
```
iverilog bug.v && ./a.out | head
F
F
F
...
```

martinwhitaker, 2018-09-23:
Icarus is behaving correctly here. The right hand side of your assignment to c is an unsigned expression (the true branch is signed but the false branch is unsigned, so the overall expression is unsigned). The Verilog >>> operator only performs an arithmetic shift if the result type is signed (that's daft, but it's what the IEEE standard says).

To get the behaviour you expect, you need to write
````
  c=1?$signed(a)>>>1:$signed(a)>>1;
````

tommythorn, 2018-09-23:
Wow, thanks. That is insane; what we’re they thinking.  I’ll fix my code.

closed 2018-09-23 by tommythorn