steveicarus/iverilog #210: Task body with multiple statements requires SystemVerilog

cdlscpmv, 2018-10-20:
When trying to compile with `-g2005`, I get the error `Task body with multiple statements requires SystemVerilog`. Is this statement actually true? I cannot find it in the standard. In fact, examples given in the standard contradict this statement.

martinwhitaker, 2018-10-20:
1364-2005 A.2.7:
````
task_declaration ::=
  task [ automatic ] task_identifier ;
  { task_item_declaration }
  statement_or_null
  endtask
| task [ automatic ] task_identifier ( [ task_port_list ] ) ;
  { block_item_declaration }
  statement_or_null
  endtask
````
`statement_or_null` only allows a single statement (although that can of course be a block statement).

Can you point to a contradictory example?

cdlscpmv, 2018-10-21:
Yeah, you are right. The statements in those examples are indeed wrapped into a single `begin...end` block. Sorry for making a confusion.

closed 2018-10-21 by cdlscpmv