steveicarus/iverilog #211: VPI search path environment variable

alexforencich, 2018-10-24:
It would be extremely useful if iverilog would support a PATH/LD_LIBRARY_PATH style environment variable for setting a search path for loading VPI modules.  AFAIK, the only way to do this right now is via command line parameters, which can be difficult to support for things like continuous integration testing when root access is not available (i.e. travis-ci) and non-standard VPI modules are required (i.e. MyHDL).

martinwhitaker, 2018-10-25:
Presumably you are running vvp from some sort of shell. Why can't you use that to expand the environment variable and pass it as an argument to vvp?

alexforencich, 2018-10-25:
vvp is being called from within myhdl as a cosimulation.  So yes, I suppose I could add some extra code to every single one of my testbench scripts to grab an environment var and pass it through to vvp as a command line arg, but this seems like a highly logical feature to configure via an environment variable directly.  You do have IVERILOG_DUMPER already which can disable dumping, and there is precedent with PATH and LD_LIBRARY_PATH.

martinwhitaker, 2018-10-25:
You could create a wrapper script around vvp to do the job. But yes, this is a perfectly reasonable request - just needs someone to find the time to do it...

alexforencich, 2018-10-26:
After taking a look at the code, this seems like a pretty simple thing to implement.  Also, it seems like the path search is backwards (a path added with -M is searched after the built-in paths instead of before so you can't override modules with -M as what might be expected).  I'd be happy to make the change, I'd just like to know what to call the variable.  IVERILOG_VPI_PATH?  VPI_MODULE_PATH?  IVERILOG_VPI_MODULE_PATH?  Something else?

martinwhitaker, 2018-10-26:
I'd go for IVERILOG_VPI_MODULE_PATH. Don't forget to update the man page.

steveicarus, 2018-10-26:
2nd it. Iverilog stuff should have iverilog in the name.

alexforencich, 2018-10-27:
See https://github.com/steveicarus/iverilog/pull/213.

martinwhitaker, 2018-10-29:
#213 merged.

closed 2018-10-29 by martinwhitaker