steveicarus/iverilog #212: Environment variable to disable dumping

alexforencich, 2018-10-24:
For continuous integration testing, dump files are not really necessary.  It would be very convenient to be able to set an environment variable that disables all dump file generation to save disk space and processor cycles while running automated tests.

martinwhitaker, 2018-10-25:
This already exists - see IVERILOG_DUMPER in the vvp man page.

closed 2018-10-25 by martinwhitaker

alexforencich, 2018-10-25:
Apologies; I didn't see that.  I will definitely be making use of that.

alexforencich, 2018-10-27:
Actually, on further inspection, this needs to be revisited.  I use the lxt2 output format for debugging as it is smaller on disk, especially with long traces.  The only way to set the output format to something other than VCD is to pass a command line argument to vvp.  So all of my testbench wrappers do this.  However, these command line arguments override this environment variable.  So, either a way to override the command line arg is required, or a different way of setting the output format is required because with the way things are set up right now, it is impossible to both use an output format other than VCD and to disable dumping via an environment variable.  

I vote for finding another way to set the output format, such as by detecting the format based on the file extension passed to $dumpfile as right now it is very odd to specify the dump file within the source but specify its format via the command line.

martinwhitaker, 2018-10-27:
Why not set IVERILOG_DUMPER=lxt2 when debugging and IVERILOG_DUMPER=none when running CI?

alexforencich, 2018-10-27:
Why not detect the format from the dump file extension?  This avoids writing out VCD files with a .lxt extension.

alexforencich, 2018-10-27:
Maybe I'll just make a separate issue for that, I think that makes the most sense.

martinwhitaker, 2018-10-27:
I was just responding to your comment that "The only way to set the output format to something other than VCD is to pass a command line argument to vvp".

Yes, please open a separate issue for auto-detecting the dump format. Patches welcome!