steveicarus/iverilog #214: Detect dump file format from dump file extension

alexforencich, 2018-10-27:
Currently, the dump file name is specified in the HDL with $dumpfile but the dump file format must be specified either with an environment variable IVERILOG_DUMPER or with a command line argument.  This is silly because if you forget to do that, then you can end up writing out VCD files (the default format) with .lxt extensions.  Additionally, the command line arguments override the IVERILOG_DUMPER environment variable, so if you set the format with a command line argument, it is impossible to override it with IVERILOG_DUMPER.  The most straightforward solution to this is to have Icarus Verilog check the extension of the file name passed to $dumpfile and set the dump file format based on that, then support overriding with IVERILOG_DUMPER and then command line arguments.

steveicarus, 2019-10-30:
Perhaps turn the problem around, and have the dumper add a suffix based on the dump format chosen?

steveicarus, 2019-10-30:
As a compromise solution, I've pushed to the master (devel) branch support for selecting the dump file suffix based on the dump type. If you give a file name without a suffix, the dumper will choose an appropriate one. If you do include a suffix, it will leave it. The thinking here is with this update, the user can enable dynamic suffix choice simply by not including a suffix, thus leaving the choice of dumper as a run-time thing.

It already chose a file name with the appropriate suffix if you did not invoke $dumpfile explicitly.

closed 2019-10-30 by steveicarus