steveicarus/iverilog #215: unable to compile in mac

peterremote1980, 2018-11-02:
[config.log](https://github.com/steveicarus/iverilog/files/2542262/config.log)

```
/Users/peter/Downloads/iverilog>gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/c++/4.2.1
Apple LLVM version 9.0.0 (clang-900.0.38)
Target: x86_64-apple-darwin18.0.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

```
/Users/peter/Downloads/iverilog>g++ --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/c++/4.2.1
Apple LLVM version 9.0.0 (clang-900.0.38)
Target: x86_64-apple-darwin18.0.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

```
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stack:86:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/deque:2335:9: error: too many template arguments for class template 'auto_ptr'
        unique_ptr<pointer, _Dp> __hold(
        ^                   ~~~~
./pform_types.h:37:20: note: expanded from macro 'unique_ptr'
#define unique_ptr auto_ptr
                   ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:2006:28: note: template is declared here
class _LIBCPP_TEMPLATE_VIS auto_ptr
                           ^
In file included from parse.y:30:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stack:86:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/deque:2478:9: error: too many template arguments for class template 'auto_ptr'
        unique_ptr<pointer, _Dp> __hold(
        ^                   ~~~~
./pform_types.h:37:20: note: expanded from macro 'unique_ptr'
#define unique_ptr auto_ptr
```

martinwhitaker, 2018-11-02:
I've pushed a possible fix for this to the master branch. Please let me know if it fixes the problem, as I can't reproduce it with the compilers I use.

martinwhitaker, 2018-12-13:
No response, so I'll assume it's fixed. Fix also pushed to the v10 branch.

closed 2018-12-13 by martinwhitaker