steveicarus/iverilog #219: SystemVerilog explicit resizing and signedness

AndreasLoow, 2018-12-03:
Consider the following Verilog file `Main.sv`:

```
module Main();

   logic[2:0] a = 3'b111;
   logic signed[2:0] a_signed = 3'b111;

   logic[2:0] b = 0;

   logic[3:0] c0;
   logic[3:0] c1;

   initial begin
      c0 = 4'($signed(a)) + b;
      c1 = 4'(a_signed) + b;

      $display("c0: %b", c0);
      $display("c1: %b", c1);

      $finish();
   end

endmodule
```

Doing `iverilog -g2012 Main.sv && ./a.out` results in

```
c0: 0111
c1: 0111
```

Is this really correct? Consider what the SystemVerilog-2017 standard (I don't have access to the 2012 version) says on p. 133:

> The signedness shall pass through unchanged, i.e., the signedness of the result shall be the self-determined signedness of the expression inside the cast.

Shouldn't both `c0` and `c1` be `1111`?

martinwhitaker, 2018-12-03:
I think you are right, and testing with a couple of other simulators shows they agree.

martinwhitaker, 2018-12-15:
Fixed in both the master and v10 branches.

closed 2018-12-15 by martinwhitaker