steveicarus/iverilog #223: starting from version 10.0 when signals are set from VPI they updated but their combination is not

vvavrychuk, 2019-01-02:
I have found that starting from v10.0 iverilog/vvp can delay updating signals combination when signals are set from VPI. This issue is not present in v0.9.7.

This is happening only in certain case. I prepared minimal reproducible example below. It is also available in https://gitlab.com/vvavrychuk/vpi-issue.

This issue makes some testbenches, for example https://github.com/ultraembedded/cores/tree/master/uart, not working on newer Icarus Verilog (https://github.com/ultraembedded/cores/issues/4).

I use following Verilog code for test
```
module tb_top;
reg clk;
reg a, b;
wire c;
assign c = a & b;

always @ (posedge clk)
  $display("a = ", a, ", ",
           "b = ", b, ", ",
           "c = ", c);
endmodule
```

In VPI code on simulation start I set initial values and schedule timer
```
static int simulation_started_cb(p_cb_data cb_data)
{
    set_value("tb_top.clk", 0);
    set_value("tb_top.a", 0);
    set_value("tb_top.b", 0);
    start_clock_timer();
}

static void startup_routine(void)
{
    s_cb_data cb_data;
    cb_data.user_data = NULL;
    cb_data.reason    = cbStartOfSimulation;
    cb_data.cb_rtn    = simulation_started_cb;
    cb_data.time      = NULL;
    cb_data.value     = NULL;
    vpi_register_cb(&cb_data);
}

void (*vlog_startup_routines[])() = 
{
    startup_routine,
    0
};
```

In timer handler I set other values
```
static int clock_timer_cb(p_cb_data cb_data)
{
    set_value("tb_top.clk", 1);
    set_value("tb_top.a", 1);
    set_value("tb_top.b", 1);
}
```

As result I get
```
a = 1, b = 1, c = 0
```
when running simulation.

Helper functions used in VPI are:
```
static int get_value(const char* name)
{
    vpiHandle handle = vpi_handle_by_name(name, NULL);
    assert(handle != NULL);

    s_vpi_value value;
    value.format = vpiIntVal;
    vpi_get_value(handle, &value);

    return value.value.integer;
}

static void set_value(const char* name, int value)
{
    vpiHandle handle = vpi_handle_by_name(name, NULL);
    assert(handle != NULL);

    s_vpi_value value_s;
    value_s.format = vpiIntVal;
    value_s.value.integer = value;

    s_vpi_time time;
    time.type = vpiSimTime;
    time.high = 0;
    time.low  = 0;

    vpi_put_value(handle, &value_s, &time, vpiInertialDelay);
}

static void start_clock_timer()
{
    s_vpi_time time;
    time.type = vpiSimTime;
    time.high = 0;
    time.low  = 1;

    s_cb_data cb_data;
    cb_data.user_data = NULL;
    cb_data.reason    = cbAfterDelay;
    cb_data.cb_rtn    = clock_timer_cb;
    cb_data.time      = &time;
    cb_data.value     = NULL;
    vpi_register_cb(&cb_data);
}
```

martinwhitaker, 2019-01-06:
I don't see that this is a bug. Setting the values of a and b adds an update event for c to the stratified event queue. Setting the value of clk triggers the @(posedge clk), which adds an evaluation event for that process to the stratified event queue. The simulator is allowed to execute those events in any order.

vvavrychuk, 2019-01-10:
@martinwhitaker sorry, I made a mistake and my above minimal reproducible example does not work both on v0.9.7 and v10.0.

I have updated my example in https://gitlab.com/vvavrychuk/vpi-issue and now on v10.0 it gives
```
vvp -m issue -M . issue.vvp
a = 1, b = 1, c = 0
```
and on v0.9.7 it gives
```
vvp -m issue -M . issue.vvp
a = 0, b = 0, c = 0
```
Maybe this changes something?

Let me explain how my minimal reproducible example is done now.

Now my clock is coming from Verilog so that `tb_top.v` looks like
```
module tb_top;
reg clk = 0;
reg a = 0, b = 0;
wire c;
assign c = a & b;

initial #1 clk = 1;

always @ (posedge clk)
  $display("a = ", a, ", ",
           "b = ", b, ", ",
           "c = ", c);
endmodule
```

Helper functions `get_value` and `set_value` are unchanged. In `vlog_startup_routines` I register for `cbStartOfSimulation` as before. In simulation started even handler I subscribe for clock change event
```
s_cb_data cb_data = {0};
cb_data.reason    = cbValueChange;
cb_data.obj       = vpi_handle_by_name("tb_top.clk", NULL);
cb_data.cb_rtn    = clock_value_changed_cb;

s_vpi_time time   = {0};
time.type         = vpiSimTime;
cb_data.time      = &time;

s_vpi_value value = {0};
value.format      = vpiIntVal;
cb_data.value     = &value;

vpi_register_cb(&cb_data);
```

In clock change event I check that I get high value and change `a` and `b` to high value
```
if (cb_data->value->value.integer != 0) {
    set_value("tb_top.a", 1);
    set_value("tb_top.b", 1);
}
```

I think behaviour of v10.0 is not correct because I use `vpiInertialDelay`.

red0bear, 2019-02-05:
hello. Can you see this https://github.com/GLADICOS/SPACEWIRESYSTEMC

steveicarus, 2019-02-05:
I think this is a conversation best had on the iverilog-devel mailing list.

vvavrychuk, 2019-02-06:
I was able to solve my problem by using system task function to advance VPI simulation
```
always @ (posedge clk)
    $clock_change;
```
Idea used in @red0bear.

It works (as expected) only when I use `vpiInertialDelay` (even zero one):
```
iverilog -o issue.vvp tb_top.v
gcc -shared -fPIC -o issue.vpi -I/usr/include/iverilog/ issue_vpi.c
vvp -m issue -M . issue.vvp
time =                    1, a = 0, b = 0, c = 0
clock_change_calltf, time = 1
clock_change_calltf, time = 3
time =                    3, a = 0, b = 0, c = 0
```
But if I use `vpiNoDelay` I get:
```
gcc -shared -fPIC -o issue.vpi -I/usr/include/iverilog/ issue_vpi.c
vvp -m issue -M . issue.vvp
time =                    1, a = 0, b = 0, c = 0
clock_change_calltf, time = 1
clock_change_calltf, time = 3
time =                    3, a = 1, b = 1, c = 0
```
Full code is in https://gitlab.com/vvavrychuk/vpi-issue/tree/advance-using-system-tf.

Still not sure that it will work forever and will not break with some Icarus Verilog version. Because as it is was explained by @martinwhitaker Icarus Verilog is free to choose between added `vpiInertialDelay` with zero delay and already pending events.

So only correct solution for me seems to use non-zero inertial delay when changing values from VPI code.

closed 2019-02-06 by vvavrychuk