steveicarus/iverilog #225: error: ‘LONG_LONG_MIN’ was not declared

balanx, 2019-01-21:
OS : WIN10, cygwin, X86_64
commit : e745304cc4256ad881e8dd41e0ec61623e965368

There is an error when do 'make':

vthread.cc: In function ‘bool of_MOD_S(vthread_t, vvp_code_t)’:
vthread.cc:4003:17: error: ‘LONG_LONG_MIN’ was not declared in this scope
      if ((lv == LONG_LONG_MIN) && (rv == -1))
                 ^~~~~~~~~~~~~
vthread.cc:4003:17: note: suggested alternative: ‘LLONG_MIN’
      if ((lv == LONG_LONG_MIN) && (rv == -1))
                 ^~~~~~~~~~~~~
                 LLONG_MIN
MAKE[1]: *** [Makefile:148: vthread.o] Error 1

martinwhitaker, 2019-01-26:
Also fixed in the v10 branch.

closed 2019-01-26 by martinwhitaker

balanx, 2019-01-31:
thanks very much for your reply