steveicarus/iverilog #226: Assertion failure on independent range assignment to vector within a struct

engr248, 2019-01-24:
The first example below causes an assertion failure that seems to be related to having independent assignments to independent ranges within a vector inside of a struct. Only the assignment to `example_a` has a problem, the other assignments to single bits and vectors within the struct are not problematic.
```
module assert_failure (
    input logic         PCLK,
    input logic         PRESETn,
    input logic         PSEL,
    input logic         PWRITE,
    input logic   [3:0] PSTRB,
    input logic   [3:0] PADDR,
    input logic  [31:0] PWDATA,
    input logic         PENABLE,
    input logic   [2:0] PPROT,
    output logic        PREADY,
    output logic [31:0] PRDATA,
    output logic        PSLVERR
);

struct packed {
    logic [14:0] example_a;
    logic example_b;
    logic example_c;
    logic example_d;
    logic example_e;
    logic example_f;
    logic example_g;
    logic [4:0] example_h;
} control;

assign PREADY = 1'b1;
assign PSLVERR = 1'b0;

always @(posedge PCLK or negedge PRESETn) begin
    if (PRESETn == 1'b0) begin
        control <= '0;
    end else begin
        // Handle Register Writes
        if (PSEL && PWRITE && PENABLE && (PADDR[3:2] == 2'b00)) begin
            //Assertion failure on the following 2 ifs
            if (PSTRB[3]) begin
                control.example_a[14:7] <= PWDATA[31:24];
            end
            if (PSTRB[2]) begin
                control.example_a[6:0] <= PWDATA[23:17];
                //bit 16 intentionally ignored
            end
            if (PSTRB[1]) begin
                control.example_b <= PWDATA[15];
                control.example_c <= PWDATA[14];
                control.example_d <= PWDATA[13];
                //bits 12:10 intentionally ignored
                control.example_e <= PWDATA[9];
                control.example_f <= PWDATA[8];
            end
            if (PSTRB[0]) begin 
                control.example_g <= PWDATA[7];
                //bits 6:5 explicitly ignored
                control.example_h <= PWDATA[4:0];
            end
        end
    end
end

endmodule: assert_failure
```
Compilation with version 10.2: `iverilog -g2012 -tnull -Wall assert_failure.sv`
Output of iverilog command: `assert_failure.sv:38: assert: elab_lval.cc:1204: failed assertion use_sel == index_component_t::SEL_NONE || use_sel == index_component_t::SEL_BIT`

The following alternative/equivalent syntax is one way to work around the assertion failure that compiles.
```
module no_failure (
    input logic         PCLK,
    input logic         PRESETn,
    input logic         PSEL,
    input logic         PWRITE,
    input logic   [3:0] PSTRB,
    input logic   [3:0] PADDR,
    input logic  [31:0] PWDATA,
    input logic         PENABLE,
    input logic   [2:0] PPROT,
    output logic        PREADY,
    output logic [31:0] PRDATA,
    output logic        PSLVERR
);

struct packed {
    logic [14:0] example_a;
    logic example_b;
    logic example_c;
    logic example_d;
    logic example_e;
    logic example_f;
    logic example_g;
    logic [4:0] example_h;
} control;

assign PREADY = 1'b1;
assign PSLVERR = 1'b0;

always @(posedge PCLK or negedge PRESETn) begin
    if (PRESETn == 1'b0) begin
        control <= '0;
    end else begin
        // Handle Register Writes
        if (PSEL && PWRITE && PENABLE && (PADDR[3:2] == 2'b00)) begin
            control.example_a <= {
                PSTRB[3] ? PWDATA[31:24] : control.example_a[14:7],
                PSTRB[2] ? PWDATA[23:17] : control.example_a[6:0]
                //bit 16 intentionally ignored
            };
            if (PSTRB[1]) begin
                control.example_b <= PWDATA[15];
                control.example_c <= PWDATA[14];
                control.example_d <= PWDATA[13];
                //bits 12:10 intentionally ignored
                control.example_e <= PWDATA[9];
                control.example_f <= PWDATA[8];
            end
            if (PSTRB[0]) begin 
                control.example_g <= PWDATA[7];
                //bits 6:5 intentionally ignored
                control.example_h <= PWDATA[4:0];
            end
        end
    end
end

endmodule: no_failure
```
Compilation with version 10.2: `iverilog -g2012 -tnull -Wall assert_failure.sv`
Output of iverilog command: (null)

martinwhitaker, 2019-01-26:
The compiler doesn't currently support assignment to a part select of a struct member. I've pushed a fix to output a suitable "sorry" message rather than causing an assertion failure.

engr248, 2019-01-26:
Thanks for the explanation! The helpful error message is definitely an improvement as well.

steveicarus, 2019-09-27:
I've pushed into git master a fix, so that this sort of code is now supported. I also updated br_gh226 in the test suite to reflect that this feature now works.

closed 2019-09-27 by steveicarus

engr248, 2019-10-17:
Cheers! Thanks!