steveicarus/iverilog #229: SystemVerilog - class extensions not supported?

svenka3, 2019-02-06:
I am trying the below code on iverilog 102. build and see error as below. Am I missing a switch or is this support not yet there in Icarus?

Thanks
Srini

--- Error --
iverilog ../ref/inherit_base.sv -g2012 
../ref/inherit_base.sv:51: assert: elab_scope.cc:512: failed assertion scope
sh: line 1: 22155 Done                    /usr/lib64/ivl/ivlpp -L -F"/tmp/ivrlg2550493e9" -f"/tmp/ivrlg550493e9" -p"/tmp/ivrli550493e9"
     22156 Aborted                 (core dumped) | /usr/lib64/ivl/ivl -C"/tmp/ivrlh550493e9" -C"/usr/lib64/ivl/vvp.conf" -

---- Code:              
 
class s2p_pkt;
 bit [15:0] hdr= 16'd10;
 bit [15:0] pkt_pld=16'd30;

  virtual function bit [31:0] bit_pack();
      bit_pack = {hdr,pkt_pld};
  endfunction : bit_pack

  function void display ();
    $display ("CVC:VSV: Lab3: Learning Inheritence");
    $display ("CVC:VSV: Lab3: Extended class %m has base properties hdr %0b pkt_pld %0b AND Extra err_pkt ", hdr, pkt_pld );
  endfunction : display

endclass : s2p_pkt

class s2p_err_pkt extends s2p_pkt;
endclass

module top;
endmodule

martinwhitaker, 2019-02-06:
There are tests in the test suite that have extended classes, so there is at least some support for that. Even if you have found some aspect that isn't yet supported, the compiler should output a proper error message, not fail an assertion.

martinwhitaker, 2019-07-24:
This is a duplicate of issue #167 (opened by you!) which was fixed in October 2017. Please do test the head of the v10 branch before reporting issues.

closed 2019-07-24 by martinwhitaker