steveicarus/iverilog #230: vpi error: vpi_get_str(vpiFullName, 0) called with null vpiHandle.

promach, 2019-02-09:
My iverilog version is Icarus Verilog version 10.2 (stable) (v10_2)

I am facing this error: _vpi error: vpi_get_str(vpiFullName, 0) called with null vpiHandle._ with the dumpvars() generate loop within test_multiply.v

> [phung@archlinux multiply]$ iverilog -o multiply test_multiply.v multiply.v
> [phung@archlinux multiply]$ vvp multiply
> VCD info: dumpfile dump.vcd opened for output.
> vpi error: vpi_get_str(vpiFullName, 0) called with null vpiHandle.
> Segmentation fault (core dumped)
> [phung@archlinux multiply]$ 

**test_multiply.v**

~~~
// Testbench
module test_multiply;

  parameter	A_WIDTH=16, B_WIDTH=16;
  
  reg i_clk;
  reg i_reset;
  reg i_ce;
  reg signed[(A_WIDTH-1):0]	i_a;
  reg signed[(B_WIDTH-1):0]	i_b;
  wire signed[(A_WIDTH+B_WIDTH-1):0]	o_p;
  wire o_valid;
  
  // Instantiate design under test
  multiply mul(.clk(i_clk), .reset(i_reset), .in_valid(i_ce), .in_A(i_a), .in_B(i_b), .out_valid(o_valid), .out_C(o_p));
  
  
  initial begin
    // Dump waves
    $dumpfile("test_multiply.vcd");
    $dumpvars(0, test_multiply);
    
    i_clk = 0;
    i_reset = 0;
    i_ce = 0;
    i_a = 0;
    i_b = 0;

  end

  genvar i, j, k; // array index
  
  generate
  	for(i = 0; i < 4; i = i + 1) begin
		for(j = 0; j < 16; j = j + 1) begin
			for(k = 0; k < 32; k = k + 1) begin
				initial $dumpvars(0, test_multiply.mul.middle_layers[i][j][k]);
			end
		end
	end
  endgenerate

  always #5 i_clk = !i_clk;
 
  initial begin
    
    @(posedge i_clk);
    @(posedge i_clk);

    $display("Reset flop.");

    i_reset = 1;

    @(posedge i_clk);
	@(posedge i_clk);
	
    i_reset = 0;

    @(posedge i_clk);
    @(posedge i_clk);

    i_ce = 1;
    i_a = 16'h8;
    i_b = 16'h7;

    #400 $finish;

  end
  
endmodule
~~~

**multiply.v**

~~~
module multiply(clk, reset, in_valid, out_valid, in_A, in_B, out_C); // C=A*B

parameter A_WIDTH = 16;
parameter B_WIDTH = 16;

input clk, reset;
input in_valid; // to signify that in_A, in_B are valid, multiplication process can start
input signed [(A_WIDTH-1):0] in_A;
input signed [(B_WIDTH-1):0] in_B;
output signed [(A_WIDTH+B_WIDTH-1):0] out_C;
output reg out_valid; // to signify that out_C is valid, multiplication finished

/* 
   This signed multiplier code architecture is a combination of row adder tree and 
   modified baugh-wooley algorithm, thus requires an area of O(N*M*logN) and time O(logN)
   with M, N being the length(bitwidth) of the multiplicand and multiplier respectively

   see https://i.imgur.com/NaqjC6G.png or 
   Row Adder Tree Multipliers in http://www.andraka.com/multipli.php or
   https://pdfs.semanticscholar.org/415c/d98dafb5c9cb358c94189927e1f3216b7494.pdf#page=10
   regarding the mechanisms within all layers

   In terms of fmax consideration: In the case of an adder tree, the adders making up the levels
   closer to the input take up real estate (remember the structure of row adder tree).  As the 
   size of the input multiplicand bitwidth grows, it becomes more and more difficult to find a
   placement that does not use long routes involving multiple switch nodes for FPGA.  The result
   is the maximum clocking speed degrades quickly as the size of the bitwidth grows.

   For signed multiplication, see also modified baugh-wooley algorithm for trick in skipping 
   sign extension (implemented as verilog example in https://www.dsprelated.com/showarticle/555.php),
   thus smaller final routed silicon area.

   https://stackoverflow.com/questions/54268192/understanding-modified-baugh-wooley-multiplication-algorithm/

   All layers are pipelined, so throughput = one result for each clock cycle 
   but each multiplication result still have latency = NUM_OF_INTERMEDIATE_LAYERS 
*/

// The multiplication of two numbers is equivalent to adding as many copies of one 
// of them, the multiplicand, as the value of the other one, the multiplier.
// Therefore, multiplicand always have the larger width compared to multipliers

localparam SMALLER_WIDTH = (A_WIDTH <= B_WIDTH) ? A_WIDTH : B_WIDTH;
localparam LARGER_WIDTH = (A_WIDTH > B_WIDTH) ? A_WIDTH : B_WIDTH;

wire [(LARGER_WIDTH-1):0] MULTIPLICAND = (A_WIDTH > B_WIDTH) ? in_A : in_B ;
wire [(SMALLER_WIDTH-1):0] MULTIPLIPLIER = (A_WIDTH <= B_WIDTH) ? in_A : in_B ;

localparam NUM_OF_INTERMEDIATE_LAYERS = $clog2(SMALLER_WIDTH);

/*Binary multiplications and additions for partial products rows*/

// first layer has "SMALLER_WIDTH" entries of data of width "LARGER_WIDTH"
// This resulted in a binary tree with faster vertical addition processes as we have 
// lesser (NUM_OF_INTERMEDIATE_LAYERS) rows to add

// intermediate partial product rows additions
// Imagine a rhombus of height of "SMALLER_WIDTH" and width of "LARGER_WIDTH"
// being re-arranged into binary row adder tree
// such that additions can be done in O(logN) time

//reg [(NUM_OF_INTERMEDIATE_LAYERS-1):0][(SMALLER_WIDTH-1):0][(A_WIDTH+B_WIDTH-1):0] middle_layers;
reg [(A_WIDTH+B_WIDTH-1):0] middle_layers[(NUM_OF_INTERMEDIATE_LAYERS-1):0][0:(SMALLER_WIDTH-1)];
//reg [(NUM_OF_INTERMEDIATE_LAYERS-1):0] middle_layers [0:(SMALLER_WIDTH-1)] [(A_WIDTH+B_WIDTH-1):0];
//reg middle_layers [(NUM_OF_INTERMEDIATE_LAYERS-1):0][0:(SMALLER_WIDTH-1)][(A_WIDTH+B_WIDTH-1):0];

generate // duplicates the leafs of the binary tree

	genvar layer; // layer 0 means the youngest leaf, layer N means the tree trunk

  	for(layer=0; layer<NUM_OF_INTERMEDIATE_LAYERS; layer=layer+1) begin: intermediate_layers

		integer pp_index; // leaf index within each layer of the tree
		integer bit_index; // index of binary string within each leaf

		always @(posedge clk)
		begin
			if(reset) 
			begin
				for(pp_index=0; pp_index<SMALLER_WIDTH ; pp_index=pp_index+1)
					middle_layers[layer][pp_index] <= 0;
			end

			else begin
			
				if(layer == 0)  // all partial products rows are in first layer
				begin
				
					// generation of partial products rows
					for(pp_index=0; pp_index<SMALLER_WIDTH ; pp_index=pp_index+1)
						middle_layers[layer][pp_index] <= 
						(MULTIPLICAND & MULTIPLIPLIER[pp_index]);
						
					// see modified baugh-wooley algorithm: https://i.imgur.com/VcgbY4g.png
					for(pp_index=0; pp_index<SMALLER_WIDTH ; pp_index=pp_index+1)
						middle_layers[layer][pp_index][LARGER_WIDTH-1] <= 
					   !middle_layers[layer][pp_index][LARGER_WIDTH-1];
						
					middle_layers[layer][SMALLER_WIDTH-1] <= !middle_layers[layer][SMALLER_WIDTH-1];
					middle_layers[layer][0][LARGER_WIDTH] <= 1;
					middle_layers[layer][SMALLER_WIDTH-1][LARGER_WIDTH] <= 1;
				end
				
				// adding the partial product rows according to row adder tree architecture
				else begin
					for(pp_index=0; pp_index<(SMALLER_WIDTH >> layer) ; pp_index=pp_index+1)
						middle_layers[layer][pp_index] <=
						middle_layers[layer-1][pp_index<<1] +
		              (middle_layers[layer-1][(pp_index<<1) + 1]) << 1;
		            
		            // bit-level additions using full adders
		            /*for(pp_index=0; pp_index<SMALLER_WIDTH ; pp_index=pp_index+1)
		            	for(bit_index=0; bit_index<(LARGER_WIDTH+layer); bit_index=bit_index+1)
							full_adder fa(.clk(clk), .reset(reset), .ain(), .bin(), .cin(), .sum(), .cout());*/
		      	end
			end
		end
	end

endgenerate

assign out_C = (reset)? 0 : middle_layers[NUM_OF_INTERMEDIATE_LAYERS-1][0];

/*Checking if the final multiplication result is ready or not*/

reg [($clog2($clog2(SMALLER_WIDTH))-1):0] out_valid_counter; // to track the multiply stages
reg multiply_had_started;

always @(posedge clk)
begin
	if(reset) 
	begin
		multiply_had_started <= 0;
		out_valid <= 0;
		out_valid_counter <= 0;
	end

	else if(out_valid_counter == $clog2(SMALLER_WIDTH)-1) begin
		multiply_had_started <= 0;
		out_valid <= 1;
		out_valid_counter <= 0;
	end
	
	else if(in_valid && !multiply_had_started) begin
		multiply_had_started <= 1;
		out_valid <= 0; // for consecutive multiplication
	end
	
	else begin
		out_valid <= 0;
		if(multiply_had_started) out_valid_counter <= out_valid_counter + 1;
	end
end

`ifdef FORMAL

initial assume(in_valid == 0);
initial assert(out_valid == 0);
initial assert(out_valid_counter == 0);

wire sign_bit = in_A[A_WIDTH-1] ^ in_B[B_WIDTH-1];

always @(posedge clk)
begin
	if(reset) assert(out_C == 0);
	
	else if(out_valid) begin
		assert(out_C == (in_A * in_B));
		assert(out_C[A_WIDTH+B_WIDTH-1] == sign_bit);
	end
end

`endif

`ifdef FORMAL

always @(posedge clk)
begin
	cover(in_valid && (in_A != 0) && (in_B != 0));
	cover(out_valid);
end

`endif

endmodule

module full_adder(clk, reset, ain, bin, cin, sum, cout);

input clk, reset;
input ain, bin, cin;
output reg sum, cout;

// Full Adder Equations
// Sum = A ⊕ B ⊕ Cin and Cout = (A ⋅ B) + (Cin ⋅ (A ⊕ B))
// where A ⊕ B is equivalent to A XOR B , A ⋅ B is equivalent to A AND B

always @(posedge clk)
begin
	if(reset)
	begin
		sum <= 0;
		cout <= 0;
	end
	
	else begin
		sum <= ain^bin^cin;
		cout <= (ain & bin) | (cin & (ain^bin));
		//cout <= (ain * bin) + (cin * (ain - bin)); 
	end
end

endmodule
~~~

martinwhitaker, 2019-02-09:
Although we should try to produce a better error message, and not segfault, this code is invalid. The argument to $dumpvars must be either a module name, or a variable name, or an array word. You are passing a bit-select expression.

promach, 2019-02-10:
> The argument to $dumpvars must be either a module name, or a variable name, or an array word. You are passing a bit-select expression.

In this case, what is wrong with `initial $dumpvars(0, test_multiply.mul.middle_layers[i][j][k]);` ?

martinwhitaker, 2019-02-10:
````
reg [(A_WIDTH+B_WIDTH-1):0] middle_layers[(NUM_OF_INTERMEDIATE_LAYERS-1):0][0:(SMALLER_WIDTH-1)];
````
From IEEE 1364-2005 section 18.2.1:

> The VCD format does not support a mechanism to dump part of a vector. For example, bits 8 to 15 ([8:15]) of a 16-bit vector cannot be dumped in VCD file; instead, the entire vector ([0:15]) has to be dumped. In addition, expressions, such as a + b, cannot be dumped in the VCD file.

The VCD format doesn't support dumping array words either, but Icarus supports this as an unofficial extension.

promach, 2019-02-11:
> The VCD format does not support a mechanism to dump part of a vector.

I suppose that in my testbench, I used three for-loop index i, j and k to loop over the entire 3D array ?

~~~
  genvar i, j, k; // array index
  
  generate
  	for(i = 0; i < 4; i = i + 1) begin
		for(j = 0; j < 16; j = j + 1) begin
			for(k = 0; k < 32; k = k + 1) begin
				initial $dumpvars(0, test_multiply.mul.middle_layers[i][j][k]);
			end
		end
	end
  endgenerate
~~~

martinwhitaker, 2019-02-11:
> I suppose that in my testbench, I used three for-loop index i, j and k to loop over the entire 3D array ?
You don't have a 3D array, you have a 2D array (of vectors). If you remove the inner loop from your code above, it will work.

martinwhitaker, 2019-07-25:
I've updated vvp to output a better error message for this.

closed 2019-07-25 by martinwhitaker