steveicarus/iverilog #231: SystemVerilog: foreach crashes iverilog

japm48, 2019-02-17:
The following code triggers an assertion error. I think it is valid SystemVerilog (as it behaves correctly in ModelSim and Aldec Riviera):
```systemverilog
module bug;

reg [4:0] a = 'b0;
initial begin
    foreach(a[i])
        $display("Value of a[%0d]=%0d",i,a[i]);
end

endmodule
```

Reproduced with:
```
$ iverilog -Wall -g2012 bug.sv -o bug
bug.sv:6: assert: elaborate.cc:4866: failed assertion dims.size() == index_vars_.size()
sh: line 1:  3219 Done                    /usr/lib/ivl/ivlpp -L -Wredef-chg -F"/tmp/ivrlg271b026f2" -f"/tmp/ivrlg71b026f2" -p"/tmp/ivrli71b026f2"
      3220 Aborted                 (core dumped) | /usr/lib/ivl/ivl -C"/tmp/ivrlh71b026f2" -C"/usr/lib/ivl/vvp.conf" -- -
$ iverilog -v
Icarus Verilog version 11.0 (devel) (s20150603-597-gdc5429e5)

[...]
```

If you need more info, please let me know.

martinwhitaker, 2019-02-18:
Looking at the compiler code, it appears that support for the foreach loop is very limited. It only supports looping over the unpacked dimensions, and all the unpacked dimensions at that.

martinwhitaker, 2019-07-25:
Fixed in both the master and v10 branches.

closed 2019-07-25 by martinwhitaker