steveicarus/iverilog #234: gperf version requirement

FelixVi, 2019-03-16:
Hi,

The readme shows that version 2.7 is to be used for gperf and I was just wondering if that is still up to date.

It seems like 2.7 was current in ~2003 and has since seen multiple updates with a current version of 3.1 as standard in Ubuntu 18.04.

As far as I can see, building with version 3.1-1 works fine, but I may not have tried to use features where things break.

If 2.7 is still a current requirement, could we add some information about where to get versions that work and which ones should be supported? Is 2.7.2 the currently recommended version?

Thanks a lot!

martinwhitaker, 2019-03-16:
I've updated the README to say "or later" for the build tool versions.

closed 2019-03-16 by martinwhitaker

lostdestiny, 2019-07-31:
I tried compiling using gperf 2.7.2. It doesn't work because gperf doesn't recognise the --ignore-case option.

martinwhitaker, 2019-08-02:
I've updated the README to say gperf 3.0 or later.