steveicarus/iverilog #235: Crash in vpi_handle_by_name

mox-mox, 2019-03-20:
I tried to verify a design containing an enum with [cocotb](https://github.com/potentialventures/cocotb). But as soon as there is an enum in the top level module and the testbench tries to access a signal in the top level module, vvp will crash.

I have created a minimal (non-)working example consisting of these three files that triggers the bug:
File 1 (`enum.sv`):
```
`default_nettype none
`timescale 1ns/1ns

	typedef enum reg[3:0]
	{
		INVALID  = 4'b0000,
		ONE      = 4'b0001,
		TWO      = 4'b0011,
		THREE    = 4'b0111,
		FOUR     = 4'b1111
	} thermometer;

module example(input wire clk);

	//   |  Uncomment either this...
	//   v
	//typedef enum reg
	//{
	//	FOO = 1'b0,
	//	BAR = 1'b1
	//} something_else;

	//   |  ... or that and the
	//   v   simulator segfaults.
	thermometer foo;

	always @(posedge clk)
	begin
		$display("Got a rising clock edge");
	end

endmodule
```

File 2 (`test_enum.py`):
```
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def test_enum(dut):
	"""
	Try enum variables
	"""
	dut._log.info("Running test!")
	for cycle in range(10):
		dut._log.info("X")
		dut.clk = 0
		dut._log.info("Y")
		yield Timer(1000)
		dut.clk = 1
		yield Timer(1000)
	dut._log.info("Running test!")
```

File 3 (`Makefile`):
```
export USER_DIR:=$(shell pwd)

include $(COCOTB)/makefiles/Makefile.inc
include $(COCOTB)/makefiles/Makefile.sim

VERILOG_SOURCES = enum.sv
TOPLEVEL = example
MODULE=test_enum
```

When running `make` -> `make debug` -> `run` -> `backtrace` in the directory containing these three files,  gdb yields the following output and backtrace:
```
> make debug
PYTHONPATH=/home/mox/projects/tests/tests_0/build/libs/x86_64:/usr/local/cocotb-1.1.0:/home/mox/projects/tests/tests_0: LD_LIBRARY_PATH=/home/mox/projects/tests/tests_0/build/libs/x86_64::/usr/lib:/usr/lib MODULE=test_enum \
        TESTCASE= TOPLEVEL=example TOPLEVEL_LANG= \
        gdb --args /usr/bin/vvp -M /home/mox/projects/tests/tests_0/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
GNU gdb (GDB) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/vvp...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/vvp -M /home/mox/projects/tests/tests_0/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:278  in embed_sim_init                  Running on Icarus Verilog version 10.2 (stable)
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:279  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
    0.00ns INFO     cocotb                                      __init__.py:132  in _initialise_testbench           Running tests with Cocotb v1.1.0 from /usr/local/cocotb-1.1.0
    0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1553100576
    0.00ns INFO     cocotb.regression                         regression.py:186  in initialise                      Found test test_enum.test_enum
    0.00ns INFO     cocotb.regression                         regression.py:320  in execute                         Running test 1/1: test_enum
    0.00ns INFO     ..oroutine.test_enum.0x7ffff5fb9908       decorators.py:209  in send                            Starting test: "test_enum"
                                                                                                                    Description:
                                                                                                                        Try enum variables

    0.00ns INFO     cocotb.example                             test_enum.py:9    in test_enum                       Running test!
    0.00ns INFO     cocotb.example                             test_enum.py:11   in test_enum                       X

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff78e1322 in __strcmp_sse2_unaligned () from /usr/lib/libc.so.6
(gdb) backtrace
#0  0x00007ffff78e1322 in __strcmp_sse2_unaligned () from /usr/lib/libc.so.6
#1  0x000055555562c69a in vpi_handle_by_name ()
#2  0x00007ffff77c6990 in VpiImpl::native_check_create (this=0x5555558a4ac0, name="clk", parent=0x5555559eb840) at VpiImpl.cpp:252
#3  0x00007ffff779fc26 in __gpi_get_handle_by_name (parent=0x5555559eb840, name="clk", skip_impl=0x0) at GpiCommon.cpp:278
#4  0x00007ffff779ff90 in gpi_get_handle_by_name (parent=0x5555559eb840, name=0x7ffff5fb9e28 "clk") at GpiCommon.cpp:325
#5  0x00007ffff61b3940 in get_handle_by_name (self=0x7ffff652cf48, args=0x7ffff5f75148) at simulatormodule.c:706
#6  0x00007ffff756aa78 in _PyMethodDef_RawFastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#7  0x00007ffff756ad11 in _PyCFunction_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#8  0x00007ffff75e18eb in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#9  0x00007ffff756a2eb in _PyFunction_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#10 0x00007ffff75e12b3 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#11 0x00007ffff7524c0b in _PyFunction_FastCallDict () from /usr/lib/libpython3.7m.so.1.0
#12 0x00007ffff7534864 in _PyObject_FastCall_Prepend () from /usr/lib/libpython3.7m.so.1.0
#13 0x00007ffff7595e50 in ?? () from /usr/lib/libpython3.7m.so.1.0
#14 0x00007ffff75d5e78 in PyObject_SetAttr () from /usr/lib/libpython3.7m.so.1.0
#15 0x00007ffff75dd4a9 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#16 0x00007ffff759765e in _PyGen_Send () from /usr/lib/libpython3.7m.so.1.0
#17 0x00007ffff756a9c8 in _PyMethodDef_RawFastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#18 0x00007ffff75960ff in _PyMethodDescr_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#19 0x00007ffff75e1479 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#20 0x00007ffff756a2eb in _PyFunction_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#21 0x00007ffff75dcc42 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#22 0x00007ffff7523b99 in _PyEval_EvalCodeWithName () from /usr/lib/libpython3.7m.so.1.0
#23 0x00007ffff756a492 in _PyFunction_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#24 0x00007ffff75dcc42 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#25 0x00007ffff756a2eb in _PyFunction_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#26 0x00007ffff75dcc42 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#27 0x00007ffff756a2eb in _PyFunction_FastCallKeywords () from /usr/lib/libpython3.7m.so.1.0
#28 0x00007ffff75dcc42 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.7m.so.1.0
#29 0x00007ffff7524c0b in _PyFunction_FastCallDict () from /usr/lib/libpython3.7m.so.1.0
#30 0x00007ffff7407eab in embed_sim_init (info=0x7fffffffd550) at gpi_embed.c:331
#31 0x00007ffff779f25a in gpi_embed_init (info=0x7fffffffd550) at GpiCommon.cpp:114
#32 0x00007ffff77cbfc9 in VpiStartupCbHdl::run_callback (this=0x5555558c98d0) at VpiCbHdl.cpp:467
#33 0x00007ffff77c7e17 in handle_vpi_callback (cb_data=0x5555559478f0) at VpiImpl.cpp:551
#34 0x0000555555625df7 in vpiStartOfSim() ()
#35 0x00005555555ff643 in schedule_simulate() ()
#36 0x00005555555bc58d in main ()
(gdb)
```

When I raised the issue in the [cocotb bug tracker](https://github.com/potentialventures/cocotb/issues/856), they noted this is probably a bug in iverilog and asked me report it. I am not sure if such a thing belongs here or in the [sourceforge bug tracker](https://sourceforge.net/p/iverilog/bugs/?page=0). Please tell me, if I should have posted there, or if you need more info.

martinwhitaker, 2019-03-23:
It's OK to report here. It would have been nice to have kept all bug reports in one place, but it's too late for that now :-(

I've reproduced this bug with a simple standalone test case. A fix will follow soon.

martinwhitaker, 2019-03-23:
Should now be fixed in both the master and v10 branches.

closed 2019-03-23 by martinwhitaker

mox-mox, 2019-03-24:
It seems the bug is gone.
Thank you very much!