steveicarus/iverilog #239: Crash in PCase::elaborate_scope

jleahy, 2019-04-25:
Hi,

I've found some (invalid) input that causes icarus verilog to crash.

This is the code:

```
module a();
b b();
endmodule
```
(in `a.v`)

```
module b();
reg x;
always @(*) begin
    if (1'b0) begin
        case (1'b0)
        1'b0: begin
        // end
        endcase
    end
end
always @(*) begin
    if (1'b0) begin
        case (1'b0)
        1'b0: begin
            if (1'b0) begin
                x <= 1'b0;
            end
        end
        1'b1: begin
        end
        endcase
    end
end
endmodule
```
(in `b.v`)

Run with `iverilog -s a a.v -y .`. This is using iverilog 10.2.

The crash occurs in `PCase::elaborate_scope` in `elab_scope.cc`. The variable `sp` is an invalid pointer.

martinwhitaker, 2019-04-27:
The crash only occurs if the faulty module is loaded via the -y option. This is because the compiler stops before elaboration if there are any syntax errors in files that are explicitly included.

martinwhitaker, 2019-07-24:
Fixed in both master and v10 branches.

closed 2019-07-24 by martinwhitaker