steveicarus/iverilog #240: SystemVerilog - ports without type declaration not equivalent to logic

tilk, 2019-05-01:
The following program fails to compile:
```systemverilog
module test(output a, input b);
    always_comb a = b;
endmodule
```
I get the following error message:
```
test.sv:2: error: a is not a valid l-value in test.
test.sv:1:      : a is declared here as wire.
Elaboration failed
```
If I add `logic` to port declaration, everything works fine:
```systemverilog
module test(output logic a, input b);
    always_comb a = b;
endmodule
```
This is incorrect - in SystemVerilog there should be no difference between not declaring a type and declaring it as `logic`.

martinwhitaker, 2019-05-03:
That's not what the IEEE standard says. From 1800-2012 section 23.2.2.3:
> If the port kind is omitted:
> ...
>  - For output ports, the default port kind depends on how the data type is specified:
>     - If the data type is omitted or declared with the implicit_data_type syntax, the port kind shall default to a net of default net type.
>    - If the data type is declared with the explicit data_type syntax, the port kind shall default to
variable.
and further on there is the example
`module mh8 (output x); // output wire logic x`

tilk, 2019-05-03:
It looks like you are correct, and Aldec Riviera on EDA Playground agrees. Now I'm wondering why both Verilator and Yosys are fine with that. Anyway, I believe the issue can be closed.

martinwhitaker, 2019-05-04:
OK, closing this.

closed 2019-05-04 by martinwhitaker