steveicarus/iverilog #241: Import into package not supported?

mox-mox, 2019-05-07:
When I tried to compile this little code snippet from the standard (p. 746, 26.6, first example, first lines) I get a syntax error. Effectively, importing into a module seems to be unsupported.
So my question is: Is this actually supported?
```
package p1;
	int x, y;
endpackage 

package p2;
	import p1::x;
	export p1::*;
endpackage 
```
Command Line:
```
$ iverilog -g2012 imports.sv
imports.sv:6: syntax error
I give up.
```

martinwhitaker, 2019-05-08:
Importing into a module is supported, but not importing into a package. I don't know why.

martinwhitaker, 2019-10-03:
Import into a package is now supported in the master branch. Export is not yet supported.

PR #863 (2023-01-29): Add support for package export
By default an identifier that has been imported into a package is not
available for imports by other packages. Only imports that have been
exported can be imported again. E.g.

```SystemVerilog
package P1;
  int x;
endpackage

package P2;
  import P1::x;
  export P1::x;
endpackage

module test;
  import P2::x; // This will only work if x has been exported.
endmodule
```

Exports follow the same syntax as imports and allow both export of specific
identifiers or wildcard export. Export supports the special `*::*` target,
which will export all imported items.

Add s [... cut]
closed 2023-01-29 by steveicarus