steveicarus/iverilog #243: Assertion Failed Error Trying to compile SV files with Icarus

mickey4u, 2019-05-09:
I tried compiling a mux.sv design module with its testbench using Icarus but I am getting the following error ```vvp: array.cc:906: void compile_var2_array(char*, char*, int, int, int, int, bool): Assertion `0' failed.
Aborted (core dumped).```

Below is a code snippet of the test bench

```
`include "mux.sv"
module tb_mux();

    bit a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7, out;
    // read
    bit expectedOutput;

    int passedFile;
    integer i;
    bit [11:0] testvector[2047:0];

    // design unit under test instantiation
    mux dut(.out(out), .a0(a0), .a1(a1), .a2(a2), .d0(d0), .d1(d1), .d2(d2), .d3(d3), .d4(d4), .d5(d5), .d6(d6), .d7(d7));
initial begin
            $readmemb("testcases", testvector);
            passedFile = $fopen("passed.txt", "w");
            if(passedFile)
                $display("file opened");
            else
                $display("couldn't open the file");
end

task assertOutput;
    begin
    int unsigned cases = 0;
    for (i = 0; i< 2048; i++)begin
        {a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7, expectedOutput} = testvector[i];

       #10;
           if(a0 == 1'b0 && a1 == 1'b0 && a2 == 1'b0 && out == expectedOutput)
                cases = cases + 1;

           else if(a0 == 1'b0 && a1 == 1'b0 && a2 == 1'b1 && out == expectedOutput)
               cases = cases + 1;

           else if(a0 == 1'b0 && a1 == 1'b1 && a2 == 1'b0 && out == expectedOutput)
               cases = cases + 1;

           else  $display(0);
    end
    $fdisplay(passedFile, "%d", cases);
end
endtask

endmodule
```
What could be wrong with my testbench for it to produce this error?

martinwhitaker, 2019-05-09:
It looks like you're using a SystemVerilog feature which isn't currently supported. Try changing your array from 'bit' to 'logic' - I think that should work.

mickey4u, 2019-05-10:
@martinwhitaker It  works but I have a warning message `$readmemb standard inconsistency following 1364-2005`.

martinwhitaker, 2019-05-10:
That's because Verilog-2001 or older specify that $readmemb will fill your array starting at index 2047 and ending at index 0, and Verilog-2005 or newer specify it will fill your array starting at index 0 and ending at index 2047. Icarus only implements the Verilog-2005 behaviour.

martinwhitaker, 2019-07-24:
Arbitrary width static bit arrays are now supported in both the master and v10 branches.

closed 2019-07-24 by martinwhitaker