steveicarus/iverilog #245: Malformed Statement

mickey4u, 2019-05-17:
I have a bunch of different implementation of mux (wrong/correct) that I am evaluating. However I keep getting this error for some of the programs.

``` ./mux.sv:4: syntax error
./mux.sv:4: error: malformed statement
```
Below is a snippet of some of these programs.
```
module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = d5;

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = d0;

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = (a0 ? d7 : d5);

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = (a2 ? d4 : d2);

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = d5;

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = (a1 ? d0 : (a1 ? d2 : (a2 ? d5 : d7)));

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = d2;

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = (a2 ? d2 : d3);

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = (a2 ? d5 : d6);

 endmodule

[... 619 lines cut ...]
out = d5;

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = d5;

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = d3;

 endmodule

module mux(output logic out, input logic a0, a1, a2, d0, d1, d2, d3, d4, d5, d6, d7);
always@(*) 

out = (a0 ? d0 : (a0 ? d4 : (a0 ? d4 : d3)));

 endmodule
```

martinwhitaker, 2019-05-17:
Please give just one example that fails to compile.

mickey4u, 2019-05-17:
@mickey4u  programs are automatically generated and run against the testbench and it is difficult for me to identify the exact one that is failing. Is there a way I can get icarus to give verbose error messages.

martinwhitaker, 2019-05-17:
Try `-v` or  `-dscopes`

mickey4u, 2019-05-22:
@martinwhitaker There is definitely something wrong. I'm generating design modules of this nature as shown below. The only difference is that the number of case statements varies. However, I keep getting 
these errors.
```
./ssdisplay.sv:6: syntax error
./ssdisplay.sv:6: syntax error
./ssdisplay.sv:6: error: malformed statement
./ssdisplay.sv:6: error: Incomprehensible case expression.
```
where line 5 and 6 are the case statements. Is the syntax wrong, because it doesn't give errors for all the modules generated.
```
module seven_segment_display(output logic[6:0] segment, input logic[3:0] bcd);
always@(bcd)
begin
case(bcd) 
 4'b1110 : segment = 7'b0100101; 
 4'b1101 : segment = 7'b1000100; 
 4'b0110 : segment = 7'b0000101; 
 default : segment = 7'b0100111; 
endcase
end
endmodule
```

martinwhitaker, 2019-05-22:
That example compiles without error for me, using either head of master branch or head of v10 branch. What OS are you using, and what version of bison?

mickey4u, 2019-05-23:
@martinwhitaker I'm using Ubuntu v18.04. I used the instruction for install Icarus verilog on ubuntu from the this link [](https://iverilog.fandom.com/wiki/Installation_Guide). There was no instruction for installing bison.

martinwhitaker, 2019-05-23:
What is the first line of output if you run `iverilog -v`?

mickey4u, 2019-05-23:
@martinwhitaker ```Icarus Verilog version 10.1 (stable) ()```

martinwhitaker, 2019-05-24:
I've built from the v10_1 tag, and still can't reproduce this bug.

mickey4u, 2019-05-24:
@martinwhitaker Finally, I have been able to locate the bug. Once in a while an invalid character was being introduced into the design modules.

martinwhitaker, 2019-05-24:
OK, closing then.

closed 2019-05-24 by martinwhitaker