steveicarus/iverilog #246: Wrong Result

mickey4u, 2019-05-20:
I have a design module and a corresponding test-bench. I am using icarus verilog as well but I am getting the number of passed cases to  be 11 instead of 5. What could be wrong? 

```
module seven_segment_display(output logic[6:0] segment, input logic[3:0] bcd);
 always@(*)
     begin
      case (bcd)
       4'b0100 : begin segment = 7'b0110011; end
       4'b0110 : begin segment = 7'b1011111; end
       4'b0001 : begin segment = 7'b0110000; end
       4'b0111 : begin segment = 7'b1110000; end
       4'b0101 : begin segment = 7'b1011011; end
       default : begin segment = 7'b0000000; end
      endcase
     end
endmodule
```

```
`include "ssdisplay.sv"
//`include "vunit_defines.svh"

module tb_ssdisplay();

    // output-segments
    logic[6:0] segment;
    logic[6:0] expectedOutput;
    // bcd
    logic[3:0] bcd;
    // test vector
    logic[10:0] testvector[15:0];
    // open passed_cases file
    int passedFile;
    int i;
    int passed_cases = 0;

    // design unit instantiation
    seven_segment_display ssd(.segment(segment), .bcd(bcd));

    initial begin
        $readmemb("testcases", testvector, 0, 15);
        // open file
        passedFile = $fopen("passed.txt", "w");
        if(!passedFile) $display("error");
        assertOutput();
        // close file
        $fclose(passedFile);
    end

/*     `TEST_SUITE begin
         `TEST_SUITE_SETUP begin
            $readmemb("../../testcases", testvector, 15, 0);
            // open file
            passedFile = $fopen("../passed.txt", "w");
            if(!passedFile) $display("error");
         end

         `TEST_CASE("Testing Seven Segment Display") begin
            assertOutput();
         end

        `TEST_CASE_CLEANUP begin
            $fclose(passedFile);
        end
     end*/

    task assertOutput;
        begin
            //passed_cases = 0;
            for( i = 0; i < 16; i++)begin
                {bcd, expectedOutput} = testvector[i];
                #10;
                if(bcd == 4'b0000 && segment == 7'b1111110) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b0001 && segment == 7'b0110000) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b0010 && segment == 7'b1101101) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b0011 && segment == 7'b1111001) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b0100 && segment == 7'b0110011) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b0101 && segment == 7'b1011011) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b0110 && segment == 7'b1011111) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b0111 && segment == 7'b1110000) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1000 && segment == 7'b1111111) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1001 && segment == 7'b1111011) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1010 && segment == 7'b0000000) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1011 && segment == 7'b0000000) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1100 && segment == 7'b0000000) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1101 && segment == 7'b0000000) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1110 && segment == 7'b0000000) begin passed_cases = passed_cases + 1; end
                else if(bcd == 4'b1111 && segment == 7'b0000000) begin passed_cases = passed_cases + 1; end
                else begin passed_cases = passed_cases + 0; end
            end
            // write passed_cases to file
            $fdisplay(passedFile, "%d", passed_cases);
        end
    endtask
endmodule
```

caryr, 2019-05-21:
Without testing: the default statement gives the correct values for bcd 4'b1010 to bcd 4'b1111 so that should be the extra six passes. When debugging something small like this adding some debugging print statements would help you determine which tests were passing and which were failing.
Cary

mickey4u, 2019-05-21:
@caryr Thanks but what about this one giving 9 without the default case statement, this program gives 9 test cases passed

```
module seven_segment_display(output logic[6:0] segment, input logic[3:0] bcd);
    always@(*)
        begin
            case (bcd)
                4'b0011 : begin segment = 7'b1011011; end
                4'b1000 : begin segment = 7'b1111011; end
                4'b1010 : begin segment = 7'b0000000; end
                4'b0000 : begin segment = 7'b1111110; end
            endcase
        end
endmodule
```
I printed out the values. It looks as if when segment is not assigned a value it is by default set to o (7'b0000000). Is that how it is suppose to work, I was expecting an unknown value.

martinwhitaker, 2019-07-23:
Have you resolved this yet? Cary's answer should have helped you.

mickey4u, 2019-07-23:
Not resolved yet.

Regards,
Michael

martinwhitaker, 2019-07-23:
Add the line
````
$display("%d %b %b %d", i, bcd, segment, passed_cases);
````
at the end of the for loop in your `assertOutput` task and see what that shows you.

steveicarus, 2019-10-05:
Without a complete example (you do not include the contents of your testcases file) we are left to assume that the problem is with your code. We cannot be expected to debug your code for you, so I'm going to close this issue. If you provide the missing parts of the issue, and offer some evidence that Icarus Verilog is incorrect, we'll be happy to reopen the issue.

closed 2019-10-05 by steveicarus