steveicarus/iverilog #247: mv: cannot stat 'parse.hh': No such file or directory

fungb, 2019-05-31:
the following error comes up upon make:

mv parse.cc.h parse.h 2>/dev/null || mv parse.hh parse.h
mv: cannot stat 'parse.hh': No such file or directory
Makefile:259: recipe for target 'parse.h' failed
make: *** [parse.h] Error 1

fungb, 2019-05-31:
>

martinwhitaker, 2019-05-31:
parse.hh should have been automatically generated by bison. Check for errors in the preceding call to bison.
No problem here, using bison 3.3.2

martinwhitaker, 2019-06-02:
A change in bison was causing this error. I've pushed a fix for this to both the master and the v10 branch. Please test if this fixes the problem for you.

fungb, 2019-06-03:
confirmed working for /master.

closed 2019-06-03 by fungb

cr1901, 2020-01-29:
One thing I've noticed is that if you need to for _any_ reason need to _rebuild_ some files (for instance, I was trying to diagnose problems generating PDF files), running `make` again will cause the build to fail with the error described in the issue.

The workaround I've resorted to for now is to remove all `parse.c` files from the build tree every time I need to rebuild to force regeneration of `parse.cc.h`. For whatever reason `parse.hh` is never generated.